noble/bleno #179: Updates not received when calling updateValueCallback repeatedly

joakimahle, 2016-03-16:
Hi, thanks for an awesome library! Im experiencing an issue at the moment but Im not sure whether its in Android or Bleno. (Im communicating with Bleno from an Android device)

Im using Bleno like a Socket (One characteristic and write data back and forth on it). The data can be bigger than the notification size limit, so Im splitting it into chunks (if neccessary) and sending `this.updateValueCallback(chunk);` for each chunk. 

Everything works as expected with up to exactly 9 chunks. When adding a 10th Chunk, I don't receive anything on my Android device. I experimented with adding `Sleep()` calls between sending each Chunk. With short sleeps (like 5 ms) there is no difference. With longer sleeps (1 second) it starts to work again (I receive the characteristic updates on Android).

So Im wondering if this is an issue with Bleno or with Android. If its an Android issue, feel free to ignore it :)

joakimahle, 2016-03-16:
I have confirmed that the problem is not on Android by implementing my Client for iOS as well. (However it could still be a limitation in the architecture of BLE and not in Bleno)

joakimahle, 2016-03-17:
Another update on the issue: The delay needed between each notification depens on how many Chunks I send. For example: 10 packets required roughly 6 ms delay to arrive correctly, 30 packets required roughly 10 ms delay etc. Still don't know if it's a Bleno issue or not.

sandeepmistry, 2016-03-20:
@joakimahle please provide a minimal code sample on how to reproduce the issue, as well as an HCI  dump capture: ```sudo hcidump -t -x```

sandeepmistry, 2016-04-03:
I'm closing this due to lack of feedback, please re-open with requested info if you are still interested.

closed 2016-04-03 by sandeepmistry

perusworld, 2016-04-18:
Here is the output from a similar usecase for me

2016-04-18 18:48:42.837331 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xeb 0x61 0x70 0x70 0x72 0x6f 0x76 0x65 0x2c 0x58 0x33 0x5a 0x6c 0x62 0x6d 0x52 0x70 0x62 0xfe 0xff
2016-04-18 18:48:42.840094 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x6d 0x64 0x66 0x64 0x58 0x56 0x70 0x5a 0x46 0x39 0x38 0x4d 0x57 0x51 0x31 0x4d 0x6d 0xfe 0xff
2016-04-18 18:48:42.842924 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x5a 0x6a 0x59 0x6a 0x41 0x74 0x4d 0x44 0x55 0x35 0x4e 0x69 0x30 0x78 0x4d 0x57 0x55 0xfe 0xff
2016-04-18 18:48:42.845366 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x32 0x4c 0x54 0x67 0x33 0x5a 0x6a 0x4d 0x74 0x4e 0x6d 0x49 0x32 0x4f 0x54 0x4e 0x6c 0xfe 0xff
2016-04-18 18:48:42.848352 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x4d 0x54 0x51 0x78 0x4e 0x7a 0x6b 0x30 0x66 0x44 0x49 0x77 0x4d 0x54 0x59 0x74 0x4d 0xfe 0xff
2016-04-18 18:48:42.851158 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x44 0x51 0x74 0x4d 0x54 0x68 0x55 0x4d 0x54 0x67 0x36 0x4e 0x44 0x67 0x36 0x4d 0x54 0xfe 0xff
2016-04-18 18:48:42.853669 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x63 0x75 0x4e 0x6a 0x55 0x7a 0x57 0x6e 0x77 0x31 0x4d 0x44 0x42 0x38 0x59 0x54 0x6b 0xfe 0xff
2016-04-18 18:48:42.856360 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x77 0x4d 0x44 0x4a 0x6a 0x59 0x7a 0x6c 0x68 0x5a 0x6a 0x64 0x69 0x4e 0x57 0x59 0x30 0xfe 0xff
2016-04-18 18:48:42.858930 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x4d 0x6e 0x77 0x32 0x5a 0x44 0x6b 0x77 0x59 0x6a 0x6b 0x79 0x59 0x6d 0x55 0x79 0x4e 0xfe 0xff
2016-04-18 18:48:42.861487 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x6a 0x55 0x79 0x59 0x6a 0x4d 0x31 0x59 0x6d 0x59 0x31 0x5a 0x57 0x4d 0x78 0x5a 0x57 0xfe 0xff
2016-04-18 18:48:42.863937 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x51 0x35 0x4d 0x44 0x6b 0x32 0x59 0x6a 0x49 0x33 0x4d 0x7a 0x59 0x30 0x5a 0x6a 0x4d 0xfe 0xff
2016-04-18 18:48:42.866449 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x77 0x5a 0x57 0x49 0x79 0x4e 0x44 0x4a 0x6c 0x5a 0x54 0x51 0x78 0x4e 0x57 0x49 0x35 0xfe 0xff
2016-04-18 18:48:42.868861 < ACL data: handle 67 flags 0x00 dlen 27
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xec 0x59 0x7a 0x6c 0x6b 0x4e 0x44 0x59 0x34 0x59 0x7a 0x59 0x35 0x59 0x6d 0x59 0x78 0x5a 0xfe 0xff
2016-04-18 18:48:42.877938 < ACL data: handle 67 flags 0x00 dlen 13
    ATT: Handle notify (0x1b)
      handle 0x000e
      value 0xed 0x54 0x49 0x33 0xfe 0xff
device: disconnected

The device just goes to on -> stateChange: poweredOff when I try to send multiple chunks in succession

sandeepmistry, 2016-04-24:
@joakimahle you're really pushing out data without delays: 18:48:42.868861 -> 18:48:42.877938

Depending on what connection interval is negotiated, this seems like expected behaviour. I suggest spacing out your notify's.

rpendleton, 2016-05-03:
@sandeepmistry What's the best way to determine the necessary interval? I'm running into this use-case as well. I need to send a large amount of data back and forth the peripheral and central, but don't know how to long I should sleep between notify's. (And unfortunately, if I don't sleep for long enough, I start running into #146.)

sandeepmistry, 2016-05-09:
@rpendleton there's currently no programatic hook to determine the connection interval. You can use `hcidump` on Linux or PacketLogger from Xcode Hardware IO tools on OS X to determine it.

Another option is to use `indicate` instead of `notify`, then there is a callback for each ack received.

farfromrefug, 2016-05-12:
I solved that issue by using a ```toSend``` stack. I then use ```onIndicate``` (which is sent when a packaged as been successfully received) to check if toSend is empty. If not i call ```updateValueCallback``` and ```slice``` the stack

No more need for a sleep timer

rpendleton, 2016-05-12:
@farfromrefug That's what I originally tried doing, but for some reason, only my first indication would be delivered. I'm not sure if the problem was on the peripheral or central side though.. I'll probably have to look into it more.

farfromrefug, 2016-05-13:
@rpendleton yes it should work fine. Although i am now facing more issues, something somewhere "kills" the bluetooth server (running bleno on a pi). Will investigate and report here.

sandeepmistry, 2016-05-15:
@farfromrefug cool, thanks for the update!

farfromrefug, 2016-05-16:
@sandeepmistry i managed to now have a very stable bluetooth connection. The issue was the mtu size.
I i stick with a 20b mtu everything works fine and i never loose the connection.

sandeepmistry, 2016-05-24:
@farfromrefug nice! Thank you for sharing this!

chrisbartley, 2018-12-18:
On Mac OS 10.13, I'm seeing onIndicate() fire within 1 millisecond after calling updateValueCallback. That seems not right, if onIndicate() is really supposed to get called only after receipt is confirmed.  When chunking a large message into 20 byte chunks, only the first 7 or so go through, and then some just get dropped. Are people really able to do chunked sending of messages with no waits?  I'm finding I can reliably send if I do a 10ms wait after the onIndicate().  If I shorten to 5ms, then it works most of the time, but definitely not all the time. Anyone have working code to share?  I'll see if I can provide a minimal version which illustrates what I'm seeing.  Thanks!

chrisbartley, 2018-12-18:
Update: I tried sending a 90KB message, chunked into 20 byte pieces, and found that 10 ms is still too fast, resulting in dozens of lost chunks.  A 50 ms wait between chunks worked, however.

chrisbartley, 2018-12-19:
Another update: running under Raspbian on a Raspberry Pi 3 B+, I don't need any wait at all.  Using indicate does the right thing so that onIndicate() is really only getting called once the chunk is successfully sent.  Is the faulty behavior a known issue on the Mac?