noble/bleno #180: Repeatable disconnection a few seconds after connection (stock examples)

WayneKeenan, 2016-03-17:
Hi,

After about 4 seconds after connecting the connection is dropped.  
The issue occurs for the battery-service and echo examples on a stock Jessie (2016-02-26) and an updated kernel, via rpi-update. 

I'm using LightBlue on iOS and Mac to test and the same occurs to both clients.
It's possible to browse the characteristics meta-info in LightBlue before the connection drops.

Versions (post rbi-update):

Raspberry Pi 3 
Node v4.4.0
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

To replicate my setup:

download: 

https://downloads.raspberrypi.org/raspbian/images/raspbian-2016-02-29/2016-02-26-raspbian-jessie.zip

(sudo raspi-config expand file system)
sudo apt-get remove nodered nodejs nodejs-legacy npm
sudo apt-get update
curl -sL https://deb.nodesource.com/setup_4.x | sudo bash -
apt-get install nodejs
sudo apt-get install -y build-essential python-dev python-rpi.gpio nodejs libudev-dev libusb-1.0-0.dev
sudo rpi-update f406502f5628d32e6ca5dadac34ff7ca59f8e27f
[REBOOT]
sudo systemctl stop bluetooth
sudo hciconfig hci0 up

[run the blend samples]

Node Debug logs attached.

[EchoService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177486/EchoService-NodeDebug.txt)
[BatterService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177487/BatterService-NodeDebug.txt)

WayneKeenan, 2016-03-17:
Interesting...

Using the same SD cardin a Pi 2 but using a Belikin BLE USB, as there is no built in BLE hardware as there is on the Pi3, the examples work ok.

I've attached the Pi2 Node logs from the examples (they show connection and evidence of then navigation of the service and characteristics in LightBlue)

I've also included the lsusb -v dump of the dongle and the dimes of the Pi2 and the Pi3

[EchoService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177563/EchoService-NodeDebug.txt)
[BatterService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177564/BatterService-NodeDebug.txt)
[pi3-dmesg.txt](https://github.com/sandeepmistry/bleno/files/177620/pi3-dmesg.txt)
[pi2-dmesg.txt](https://github.com/sandeepmistry/bleno/files/177621/pi2-dmesg.txt)
[BelkinUSB-Info.txt](https://github.com/sandeepmistry/bleno/files/177622/BelkinUSB-Info.txt)

WayneKeenan, 2016-03-17:
and to complete the circle, plugging the Belkin USB BLE dongle into the Pi3 results in the example working.   (node debug of this working case is attached for reference, along with the additional dmesg output when the dongle was plugged in)

So, the bleno examples not working on a Pi3 with it;s builtin BLE h/w is pointing at something being wrong with the kernel driver, maybe...but....

Could it be because the built-in BLE h/w on the Pi 3 (hci0) is on the 'UART' bus as opposed to the 'USB' bus the welkin adapter is (hci1): 

hci1:	Type: BR/EDR  Bus: USB
	BD Address: 5C:F3:70:68:DB:AF  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1308 acl:22 sco:0 events:84 errors:0
	TX bytes:1174 acl:22 sco:0 commands:64 errors:0

hci0:	Type: BR/EDR  Bus: UART
	BD Address: B8:27:EB:0A:F7:8E  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:1787 acl:9 sco:0 events:108 errors:0
	TX bytes:2700 acl:9 sco:0 commands:99 errors:0

[EchoService-NodeDebug-Pi3-BelkinDongle.txt](https://github.com/sandeepmistry/bleno/files/177629/EchoService-NodeDebug-Pi3-BelkinDongle.txt)

ags131, 2016-03-17:
I can also confirm this behavior on my Pi3s

sandeepmistry, 2016-03-20:
@WayneKeenan @ags131 how does this behaviour compare to ```bluetoothd```'s built-in GATT server?

Also, it would be good to compare the HCI dump capture of both: ```sudo hcidump -t -x```

WayneKeenan, 2016-03-21:
Here's the hcidump log of staring and then connecting.  I'll follow up with the bluez.
Thanks
Wayne

[hci.log.txt](https://github.com/sandeepmistry/bleno/files/181848/hci.log.txt)

WayneKeenan, 2016-03-23:
I've not had much luck getting the Bluetoothd gattserver to run in the time I had, but I did rerun with hcidump for both the USB bus and UART bus type adapters again.     

For reference, both the adapters are attached at the same time, but only the one under test has brought up using hciconfig.

[hcidump_UART.txt](https://github.com/sandeepmistry/bleno/files/185731/hcidump_UART.txt)
[hcidump_USB.txt](https://github.com/sandeepmistry/bleno/files/185732/hcidump_USB.txt)

sandeepmistry, 2016-03-23:
I've published v0.4.0 to npm, which depends on the latest ```bluetooth-hci-socket``` (v0.4.3).

It has a workaround for the kernel disconnect issue (https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/4bcafd6fc483394defd5767cf4f3f15339f49a90).

Please try it out and let me know if there are any issues.

closed 2016-03-23 by sandeepmistry

WayneKeenan, 2016-03-24:
initial touch testing looks good, the connection isn't dropping like before. thank you.

vdharmon, 2016-04-04:
I'm still seeing the connection drop after the fix.

I've pulled down the code version bluetooth-hci-socket (v0.4.3).  When I look at the code that was pulled down it doesn't have the code fix you described above.  I see the code fix in the 'UART' branch but not in the 'Master' branch.  

Regards,
Vince

vdharmon, 2016-04-04:
Another quick question.  Was the fix to address using the UART built-in bluetooth device.  Or is the fix to enable running a USB bluetooth device.

Regards

WayneKeenan, 2016-04-05:
The fix addressed the built-in UART bus type device.

Try updating to the same version of the linux kernel I'm using and retest:

```
sudo rpi-update 692dde0c1586f7310301379a502b9680d0c104fd
```

vdharmon, 2016-04-05:
I have 2 differences.  I'm using Jessie-Lite and an Android BLE Scanner.

The IOs device is working properly.

The Android device connection fails. 
The broadcast of the BLE is showing up on the Android.

From what I can tell looking at the hcidump I do not see a connect coming from the Android device.  The Android OS is 5.1.1.

If I plugin a USB device then both IOs and Android both work properly.

Regards.

vdharmon, 2016-04-06:
Anyone have any thoughts on the issue?  Could it be the low-level Bluetooth on the PI?

Regards.

sandeepmistry, 2016-04-09:
@vdharmon let's make sure you are running the fix. Please do a clean install of bleno, and share the output of ```npm ls```.

vdharmon, 2016-04-10:
npm install bleno
#---------------------------------------------------------------------------
bleno@0.4.0 node_modules/bleno
├── bplist-parser@0.0.6
├── debug@2.2.0 (ms@0.7.1)
└── bluetooth-hci-socket@0.4.3 (nan@2.2.1)
#---------------------------------------------------------------------------

npm ls
#---------------------------------------------------------------------------
├─┬ bleno@0.4.0
│ ├─┬ bluetooth-hci-socket@0.4.3
│ │ └── nan@2.2.1
│ ├── bplist-parser@0.0.6
│ └─┬ debug@2.2.0
│   └── ms@0.7.1
#---------------------------------------------------------------------------

I went and looked at the bluetooth-hci-socket code to make sure the line that had (&& data[7] == 0x00) matched your new code which it does.

sandeepmistry, 2016-04-11:
@vdharmon thanks. Let's move the discussion to https://github.com/sandeepmistry/bleno/issues/175 There's something Android specific.

jahnvievince, 2016-06-03:
I am using raspberry pi 3 with wearable device for getting data from it and write it into a txt file. I have created a shell script for connection to wearable device and then get data from that. But after sometime the connection is dropped. How can I identify that who is dropping connection, the raspberry pi inbuilt bluetoth or the device bluetooth.

Config -
raspberry pi 3 jessie
OS - ubuntu
kernel version - 4.4
bluez veriosn  - inbuilt (5.23)

sandeepmistry, 2016-06-06:
@jahnvievince an HCI dump should give you more info. as to what's going on: `sudo hcidump -t -x`

jahnvievince, 2016-06-06:
@sandeepmistry On writing the command sudo hcidump -t -x ,it shows:
HCI sniffer-Bluetooth packet analyzer ver 5.23   
device:hci0 snap_len:1500 filter:0xffffffff
and gets stuck at this point.Can you please help me with this...Thanks.

sandeepmistry, 2016-06-19:
@jahnvievince you'll need 2 terminal sessions. One running HCI dump the other running the Node.js app.