noble/bleno #181: Problem sending string on Rpi

projetenvol, 2016-03-19:
Hello,

Thank you very much for Bleno, it looks very cool!
I would like to transmit a string, or a float value using Bleno. I have installed it on my raspberry pi and the tests work properly. Now, I have created a characteristic and rewritten the onReadRequest function, but I don't manage to send the wanted data...

Here are some extracts from my code :

`function rotCharacteristic(line, id) {
        bleno.Characteristic.call(this, {
    uuid: id,
    properties: ['read'],
    descriptors: [
      new bleno.Descriptor({
        uuid: id,
        value: 'rotation selon l\'axe de la ligne ' + line,
      })
    ]
  });
this.line = line;
        
}

util.inherits(rotCharacteristic, bleno.Characteristic);

rotCharacteristic.prototype.onReadRequest = function(offset, callback) {
        console.log("ReadRequest line " + this.line);

        var line = this.line;
  if (offset) {
    callback(this.RESULT_ATTR_NOT_LONG, null);
  }
  else {
        console.log("Pas d'erreur, on read le file");
        fs.readFile("logx.txt", "utf8", function (err, data) {
                                if(err) {
                                        console.log(err);
                                }
                                else{
                                        var j = data.indexOf("\n", 0); // detect linebreak
                                        console.log(data.substr(0,j-1));
                                        console.log(data.substr(j+1, data.length - 1));
                                        console.log("line = " + line);
                                        if(parseInt(line) == 0){
                                                //var value = new Buffer(8);
                                                //value.writeFloatBE(parseFloat(data.substr(0, j -1)));
                                                console.log("Ligne = 0");
                                                callback(Characteristic.RESULT_SUCCESS, new Buffer('test', 'ascii'));
                                        }
                                        else if (parseInt(line) == 1){
                                                console.log("Line = 1");
                                                var value = new Buffer(4);
                                                value.writeFloatBE(parseFloat(data.substr(j+1, data.length -1)));
                                                callback(Characteristic.RESULT_SUCCESS, value);
                                        }
                                }
                        });

  }
};
`

I read the content of a file that has two lines : 
-169.575226
106.14984
and I would like to transmit this information.

Therefore, I created two rotCharacteristics with line = 0 and line = 1.
There are two issues : 

- most of the time, when I use LightBlue I only see the first of these two characteristics (I saw the second one approximately twice...) without having changed anything else than the onReadRequest function.
- when I read this characteristic using LightBlue, I have proof with the console.log() that the code is triggered, but I always get the same thing : 
ASCII : 
Hex : 0x0418
Decimal : 6148
Date : ...

Any help would be highly appreciated, I have spent hours on this without seeing much progress...

Thank you very much in advance!

Hippolyte

sandeepmistry, 2016-03-27:
@projetenvol your code is a bit confusing, and it's hard to tell whether it's an issue with bleno or you fs related code.

I suggest you just try to send a string without any fs code to make sure that works.

sandeepmistry, 2016-04-11:
I'm closing this do to lack of activity, please re-open with requested info if you are still interested.

closed 2016-04-11 by sandeepmistry