noble/bleno #182: Bleno doesn't work on Brillo (Linux and Android based IoT OS)

kenchris, 2016-03-29:
Changing this line 

```} else if (platform === 'linux') {```

to

```} else if (platform === 'linux' || platform === 'android') {```

and it works fine.

sandeepmistry, 2016-04-03:
@kenchris thanks for the heads up.

To confirm does only [this](https://github.com/sandeepmistry/bleno/blob/master/lib/bleno.js#L19) line need to change or also the [``os`` section in ``package.json``](https://github.com/sandeepmistry/bleno/blob/master/package.json#L9-L13)?

It would be great if you could submit a pull request ...

poussa, 2016-04-04:
I think we can live with the code change only but it would be better to have it also on .json file.

sandeepmistry, 2016-04-28:
Closed via #195.

closed 2016-04-28 by sandeepmistry