noble/bleno #184: Windows LIBUSB_TRANSFER_STALL when connecting

rajsite, 2016-04-06:
Running the examples\pizza\peripheral.js demo on Windows 7 I am able to start advertising and see the advertised BLE service in a mobile app.

However, if I try to connect to the service I get the following error:
```
events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: LIBUSB_TRANSFER_STALL
    at Error (native)
```
I captured a log here: [log-pizza.txt](https://github.com/sandeepmistry/bleno/files/207118/log-pizza.txt)

I initially filed an issue here (which contains additional environment information): https://github.com/don/node-eddystone-beacon/issues/38

But it looks like it was also reproducible using the pizza/peripheral example in bleno.

sandeepmistry, 2016-04-11:
@rajsite could you please trying downgrading ```hci-bluetooth-socket``` to v0.4.2, as v0.4.3 might have introduced an issue as per https://github.com/sandeepmistry/noble/issues/372#issuecomment-205303268.

rajsite, 2016-04-11:
I synced bleno/master, modified package.json to the following:
```
    "bluetooth-hci-socket": "0.4.2",
```

And the examples\pizza\peripheral.js example appears to work on my Windows 7 configuration!

The service advertises, I can connect / disconnect multiple times, and it looks like I can read GATT Characteristics.

I attached a log of the working behavior here: [log-pizza-working.txt](https://github.com/sandeepmistry/bleno/files/213308/log-pizza-working.txt)

sandeepmistry, 2016-04-11:
@rajsite thanks for trying, I've published v0.4.4 of ```bluetooth-hci-socket``` with a fix (https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/25999ec523325e43d5d900a7ae8866607f56fd9b). A fresh install should be fine now, please let me if there are any issues.

closed 2016-04-11 by sandeepmistry