noble/bleno #185: addLongTermKey failing

robotastic, 2016-04-12:
When I do a Random Pairing, the addLongTermKey is failing. When loadLongTermKeys writes the keys, I get an invalid parameter back. I did a sanity check of the possible branches that could be creating it here:
https://github.com/torvalds/linux/blob/c05c2ec96bb8b7310da1055c7b9d786a3ec6dc0c/net/bluetooth/mgmt.c#L4576
But I didn't see anything obvious. I confirm using /sys/kernel/debug/hci1/long_term_keys and nothing is loaded. Have you been able to get Long Term Keys to successfully load as part of a pairing?

From debug:
````
  mgmt writing -> 1300000026000100428fe8c5b6470200100000000000000000000025ca78999b3b3beed475554ebc42406700 Length: 38 +1ms
  mgmt on data ->02000000030013000d +1ms
````
https://github.com/pauloborges/bluez/blob/master/doc/mgmt-api.txt

As a work around I have been leaving  systemctl bluetooth enabled and just stopping the services. I then comment out the smp function from writing any ACL data. That seems to allow bluetoothd to handle all of the pairing but not get in the way.

robotastic, 2016-04-14:
I made a quick fork to help document/test this. I added a few debug messages in. In examples/echo I made the read characteristic secure. When you try to read it using LightBlue, you get prompt to pair, however in the debug message you can see that it doesn't successfully write the LTK, and the rest of the steps for the pairing fail.

https://github.com/robotastic/bleno

I am doing this on iOS 9.3.1, with a Raspberry Pi 3 with both the built-in bluetooth and a USB dongle. I have disable bluetooth using systemctl.

sandeepmistry, 2016-04-24:
I'm marking this as a duplicate of #187. (There's more discussion there).

closed 2016-04-24 by sandeepmistry