noble/bleno #186: Does bleno work for BeagleBone Black?

wkopen, 2016-04-12:
Hi Sandeep and other developers,

I got bleno working on my Raspberry Pi without any issue. So I tried to replicate the same stuff to a BeagleBone Black (BBB). I'm using the same Bluetooth dongle on both so I assume it shouldn't be the cause.

My BBB is running 3.8.13-bone79, and BlueZ version is 4.99-2. I use LightBlue to detect the BLE device, and in this case my device is called "tslink".

[image]

[image]

As you can see although we can see the device, but there's No Service. After click on it, I can see the UUID 5678 with characteristic 0x9012.

When I run the same thing on a Raspberry Pi, I got the following:

[image]
[image]

sandeepmistry, 2016-04-18:
@wkopen what's the output of ```uname -a```?

Could you please try setting ```isYocto``` to true in ```gap.js```: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gap.js#L11

sandeepmistry, 2016-05-02:
I'm closing this due to lack of activity.

closed 2016-05-02 by sandeepmistry