noble/bleno #189: CBPeripheral on iOS continue to disconnect every 11 seconds

mrigo, 2016-04-16:
Hi!
I'm implementing an ios app and a node app on a raspberry pi.
I need to send data to the raspy frequently, based the user interaction.
I'm not able to do this because of this continuos disconnections.. They seems to happens every 11 seconds..

I'm implementing nothing strange on my code..

```
- (void)startScan {
    if (!self.uuid)
        return;
    [self stopScan];
    
    if ([self.delegate respondsToSelector:@selector(mrManagerStartScan)])
        [self.delegate mrManagerStartScan];
    
    NSString *uuid = [self getUUID];
    @try {
        if (uuid.length != 0) {
            CBUUID *serviceUUID = [CBUUID UUIDWithString:uuid];
            [self.centralManager scanForPeripheralsWithServices:@[serviceUUID] options:nil];
        }
    }
    @catch (NSException *exception) {
        NSLog(@"%@", exception);
        [self stopScan];
    }
    @finally {
        
    }

// when it is powered on start looking for our peripheral
- (void)centralManagerDidUpdateState:(CBCentralManager *)central {
    if(central.state==CBCentralManagerStatePoweredOn)
    {
        [self startScan];
    }
}

// central manager discovered a peripheral
- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {
    self.peripheral = peripheral;
    [self.centralManager stopScan];
    self.peripheral.delegate = self;
    // make a connection to the peripheral
    [self.centralManager connectPeripheral:self.peripheral options:nil];
}

- (void)centralManager:(CBCentralManager *)central didDisconnectPeripheral:(nonnull CBPeripheral *)peripheral error:(nullable NSError *)error
{
    self.peripheral = nil;
    if (self.errorBlock)
        self.errorBlock(error);
    if ([self.delegate respondsToSelector:@selector(mrManagerDidDisconnect)]) {
        [self.delegate mrManagerDidDisconnect];
    }
}

// connected, start discovering services
- (void)centralManager:(CBCentralManager *)central didConnectPeripheral:(CBPeripheral *)peripheral {
    peripheral.delegate = self;
    if(peripheral.state == CBPeripheralStateConnected) {
        [peripheral discoverServices:nil];
        if ([self.delegate respondsToSelector:@selector(mrManagerDidConnect)]) {
            [self.delegate mrManagerDidConnect];
        }
    }
}

// services discovered - find their characteristics
- (void)peripheral:(CBPeripheral *)peripheral didDiscoverServices:(NSError *)error {
    for (CBService *service in peripheral.services) {
        [peripheral discoverCharacteristics:nil forService:service];
    }
}

// discovered some characteristics
- (void)peripheral:(CBPeripheral *)peripheral didDiscoverCharacteristicsForService:(CBService *)service error:(NSError *)error {
    for (CBCharacteristic *characteristic in service.characteristics) {
        if([characteristic.UUID isEqual:[CBUUID UUIDWithString:[self getUUID]]]) {
            self.mirrrCharacteristic = characteristic;
            self.connectionStatus = Connected;
        }
        //[peripheral readValueForCharacteristic:characteristic];
        [peripheral setNotifyValue:YES forCharacteristic:characteristic];
    }
}
    
}
```

I need a continuos connection to avoid the user to wait for a connection every time he must do something.

It's possible to have a continuos connection?
There's a timeout to set on ios or on the node app for the connection?

Thanks!

sandeepmistry, 2016-04-18:
@mrigo I think this is probably related to https://github.com/sandeepmistry/bleno/issues/180.

Could you please provide:

* a HCI dump: ```sudo hcidump -t -x```
* code you are running
* the output of ```npm ls```

mrigo, 2016-04-21:
Hi!
I'm trying this on my Mac before deploying on my rasp!
I haven't hcidump on mac but here the code:

I need a generic method to send long strings (not always, but maybe sometimes) so I've made

```
this.update_values = function(string_to_write) {

    var offset = 50;
    for (var i = 0; i<string_to_write.length;) {
        var string_to_send = "";
        string_to_send = string_to_write.substring(i, i+offset);
        var buffer = new Buffer(string_to_send, 'utf-8');
        global.updateValueCallback(buffer, 'utf-8');
        global.blenoCallback(global.bleno_success);
        i = i+offset;
    }
    this.send_end_trasmission_signal();
}

this.send_end_trasmission_signal = function() {
    global.updateValueCallback(new Buffer('*finedeigiochi*'), 'utf-8');
    global.blenoCallback(global.bleno_success);
}
```

where 

global.updateValueCallback => is the this.updateValueCallback of the bleno Characteristic
global.blenoCallback => is the callback passend in the method onWriteRequest : function(newData, offset, withoutResponse, callback) of the bleno Characteristic

Here the output of `npm ls`

```
project@1.0.0 /Users/Mattia/project
├─┬ bleno@0.3.5
│ ├── UNMET OPTIONAL DEPENDENCY bluetooth-hci-socket@~0.4.0
│ ├── bplist-parser@0.0.6
│ ├─┬ debug@2.2.0
│ │ └── ms@0.7.1
│ └─┬ xpc-connection@0.1.4
│   └── nan@2.2.0
├─┬ body-parser@1.15.0
│ ├── bytes@2.2.0
│ ├── content-type@1.0.1
│ ├── depd@1.1.0
│ ├─┬ http-errors@1.4.0
│ │ ├── inherits@2.0.1
│ │ └── statuses@1.2.1
│ ├── iconv-lite@0.4.13
│ ├─┬ on-finished@2.3.0
│ │ └── ee-first@1.1.1
│ ├── qs@6.1.0
│ ├─┬ raw-body@2.1.6
│ │ ├── bytes@2.3.0
│ │ └── unpipe@1.0.0
│ └─┬ type-is@1.6.12
│   └── media-typer@0.3.0
├── dblite@0.7.6
├── exec@0.2.1
├─┬ express@4.13.4
│ ├─┬ accepts@1.2.13
│ │ └── negotiator@0.5.3
│ ├── array-flatten@1.1.1
│ ├── content-disposition@0.5.1
│ ├── cookie@0.1.5
│ ├── cookie-signature@1.0.6
│ ├── escape-html@1.0.3
│ ├── etag@1.7.0
[... 362 lines cut ...]
  │   └── json3@3.2.6
  ├─┬ socket.io-client@1.4.5
  │ ├── backo2@1.0.2
  │ ├── component-bind@1.0.0
  │ ├── component-emitter@1.2.0
  │ ├─┬ engine.io-client@1.6.8
  │ │ ├── component-inherit@0.0.3
  │ │ ├── has-cors@1.1.0
  │ │ ├── parsejson@0.0.1
  │ │ ├── parseqs@0.0.2
  │ │ ├── xmlhttprequest-ssl@1.5.1
  │ │ └── yeast@0.1.2
  │ ├── indexof@0.0.1
  │ ├── object-component@0.0.3
  │ ├─┬ parseuri@0.0.4
  │ │ └─┬ better-assert@1.0.2
  │ │   └── callsite@1.0.0
  │ └── to-array@0.1.4
  └─┬ socket.io-parser@2.2.6
    ├── benchmark@1.0.0
    ├── component-emitter@1.1.2
    └── json3@3.3.2
```

Yes I know.. I need a lot of modules.. It's a complicate project :)

sandeepmistry, 2016-04-24:
@mrigo 

> global.updateValueCallback(buffer, 'utf-8');

Your example is calling ```updateValueCallback``` in a for loop without delays. You're probably pushing too much data out. I would recommend adding a delay between `updateValueCallback` callbacks somehow. Remember BLE has less bandwidth than regular Bluetooth.

mrigo, 2016-04-30:
Hey,
thanks for the reply!
I've try to add a delay (1000 and 100 milliseconds) but even without any communication the Peripheral continue to disconnect.
I connect the ios app to the node js app and wait.. 10 secs and BOOM! Disconnected!
(Sorry for the drama but in this way I can explain to you my frustration..)

sandeepmistry, 2016-05-01:
@mrigo I suggest the following

* use PacketLogger from Xcode IO tools to trace
* use an app like LightBlue on iOS to test as well

As mentioned in the read me the `disconnect` event is linux only: https://github.com/sandeepmistry/bleno#disconnect

mrigo, 2016-05-05:
Ok, reading the specific of BLE this is an expected behaviour.
Bluetooth was designed to consume very little amount of energy so it drops the connection as soon as possible.
In my case I'll send/read data, disconnect and when I need to communicate again I'll reconnect to perform some actions... Not the best user experience but  ¯\_(ツ)_/¯

Thanks for your support!
\m

sandeepmistry, 2016-05-09:
@mrigo ok, I'm closing this then.

closed 2016-05-09 by sandeepmistry