noble/bleno #190: Use with Noble as Bluetooth Repeater

mikeholler, 2016-04-19:
This is not an issue, but I couldn't find a better place to put it. I was wondering if this could be used with sandeepmistry/noble to create a Bluetooth repeater/extender of some sort. I realize the device MAC and possible device name would have to change, and also that I'd need two bluetooth adapters (one for noble and one for bleno), but is this something that would be possible?

Here's what I'm thinking:

* use noble to scan for a device to extend
* once the device is found, noble connects
* once connected, use bleno to advertise
* once client connects to bleno, forward messages from noble to bleno and vice versa
* if extended device disconnects, trigger noble disconnect
* if noble detects a disconnect, trigger bleno disconnect

sandeepmistry, 2016-04-25:
@mikeholler gitter is a good place for discussions like this: https://gitter.im/sandeepmistry/noble

It could work with multiple Bluetooth adapters or a Bluetooth 4.1 adapter. Depends what the end goal is. noble has a built-in distributed mode, see:
 * https://github.com/sandeepmistry/noble/blob/master/ws-slave.js
 * https://github.com/sandeepmistry/noble/tree/master/lib/distributed

Where one node can control others which forward events to it. Maybe this fits your use case?

sandeepmistry, 2016-05-02:
I'm closing this, but feel free to continue the discussion here.

closed 2016-05-02 by sandeepmistry

mikeholler, 2016-05-02:
@sandeepmistry, thanks for the response! I'm looking for something that is a little more complex than just forwarding from one node to another. I was thinking something like this:

```
Peripheral device <-----> bleno/noble <-----> Central module
```

Ultimately the reason for doing this is to simulate a problem that I am having in setting up automated integration tests that exercise our Android application's integration with a third-party Bluetooth 4.0 device. Specifically, I'm trying to emulate a walkaway event (out-of-range disconnect) so I can test our application's ability to gracefully disconnect from and automatically reconnect to the paired device when Android registers a loss of connection.

Basically, to do this:

```
Peripheral device <-----> bleno/noble <-----> Central module
                               ^
                               |
                               v
          Secondary protocol that can be used to trigger a "walkaway" event
          by cutting the connection to the central module and noble & peripheral
          device and bleno abruptly. bleno would then be made discoverable and
          the peripheral device would also make itself discoverable (to noble)
```

If you think this library is a good start for solving this type of problem, we'd love to develop and open-source a Bluetooth proxy that will allow one to trigger Bluetooth walkaway events at will, as it seems like something that plenty of people---myself included---would like to exist.

sandeepmistry, 2016-05-08:
@mikeholler yes, using there libraries seems like a reasonable option.

You might find https://github.com/sandeepmistry/noble/commit/60fa47e114109984fac5ca4f92d7bff1604566af of interest.