noble/bleno #191: How to connect a windows 10 device with Bleno

Namarcil, 2016-04-21:
Hello there,

I want to discover services with Android, iOS and Windows 10.
Android and iOS are working perfectly, but I have issues with Windows 10.

I can see the device, but I can't discover services. As I saw, the W10 device needs to be paired to the BLE device do that.
I tried almost every possible way to pair the device with the server (custom or system pairing) but no result at the time. Do you know if it's possible to accept the pairing request from the Bleno Server ?

I'm working with a Raspberry Pi with an Asus USB-BT400 and Bleno.  

You can find the debug log of Android and W10 discovering services on this gist
https://gist.github.com/Namarcil/f8aff6fd24012b8445304ee43d631270

Hope it can help.

Thank you in advance.

sandeepmistry, 2016-04-25:
@Namarcil bleno has "experimental" support for pairing on Linux. Depending on the kernel version and BlueZ version it is know to work.

```
 mgmt on data ->110000000800101e24e2b7f40105 +1ms
  hci onSocketData: 040f0400010604 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 15 +1ms
  hci onSocketData: 04050400400013 +115ms
  hci   event type = 4 +0ms
  hci   sub event type = 5 +1ms
  hci           handle = 64 +0ms
  hci           reason = 19 +1ms
  bleno disconnectf4:b7:e2:24:1e:10 +1ms
```

There is a mgmt event in your log. Could you please provide a HCI dump capture as well. ```sudo hcidump -t -x```. What tool or steps are you using on Windows 10?

Namarcil, 2016-04-25:
Hi @sandeepmistry, thank you for your quick reply,

I am working on a 4.1.18-v7+ Linux with 5.23-2+rpi1 BlueZ

You can find on this gist the hcidump log
https://gist.github.com/Namarcil/753649da676dc32d2ad7771c64155a00

It corresponds to an attempt to pair with the build-in Windows Bluetooth tool.

sandeepmistry, 2016-05-01:
@Namarcil thanks,

```
2016-04-25 09:06:10.853032 > ACL data: handle 64 flags 0x02 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x04 oob 0x00 auth req 0x05
      max key size 0x10 init key dist 0x04 resp key dist 0x03
      Capability: KeyboardDisplay (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:    CSRK
      Responder Key Distribution:  LTK IRK
2016-04-25 09:06:10.853270 < ACL data: handle 64 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x05
      max key size 0x10 init key dist 0x04 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:    CSRK
      Responder Key Distribution:  LTK
2016-04-25 09:06:10.859572 < ACL data: handle 64 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:
      Responder Key Distribution:  LTK
2016-04-25 09:06:11.033196 > ACL data: handle 64 flags 0x02 dlen 6
    SMP: Pairing Failed (0x05)
      reason 0x07
      Reason Command Not Supported
```

It looks like the built-in SMP stack in bleno, does not support what Windows expects.

Namarcil, 2016-05-04:
@sandeepmistry Thank you for your feedback.

Do you have any idea how we could make it work ?

sandeepmistry, 2016-05-09:
@Namarcil yes, we'd need to add the relevant features to [smp.js](https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/smp.js).

The tricker item is where to persist the LTK's between app restarts.

Namarcil, 2016-05-17:
As always, thank you @sandeepmistry for your answer.

We're not sure to understand what W10 expects. 
We can see that Bleno is sending

```
new Buffer([
SMP_PAIRING_RESPONSE,
0x03, // IO capability: NoInputNoOutput
0x00, // OOB data: Authentication data not present
0x01, // Authentication requirement: Bonding - No MITM
0x10, // Max encryption key size
0x00, // Initiator key distribution: <none>
0x01 // Responder key distribution: EncKey
])
```

as response to W10 Pairing Request

Do you have any idea what part of the response is not supported by W10 ?

sandeepmistry, 2016-06-19:
@Namarcil probably something that maps to the initial request it in sending:

```
2016-04-25 09:06:10.853032 > ACL data: handle 64 flags 0x02 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x04 oob 0x00 auth req 0x05
      max key size 0x10 init key dist 0x04 resp key dist 0x03
      Capability: KeyboardDisplay (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:    CSRK
      Responder Key Distribution:  LTK IRK
```

(see the HCI dump capture you send in https://github.com/sandeepmistry/bleno/issues/191#issuecomment-214238966).

sandeepmistry, 2016-09-26:
@Namarcil any progress on this?

Namarcil, 2016-09-26:
Hi @sandeepmistry 

To be honnest, my previous company kinda drop the project, and I'm no longer part of it, so if you need the contacts I can pm you

Thanks again for your help

sandeepmistry, 2016-09-26:
Hi @Namarcil,

No worries, let's close this for now then. It can be re-opened when there is interest again.

closed 2016-09-26 by sandeepmistry