noble/bleno #192: Module version mismatch error

shekit, 2016-04-22:
Hi, 

I'm trying to install and use bleno in an electron app, however I keep getting this error in the console when I require bleno

`Uncaught Error: Module version mismatch. Expected 47, got 14.`

When I npm install bleno, this is what shows up in the console:

`xpc-connection@0.1.4 install /Users/abhishek/Google Drive/startups/thesis/code/electron/node_modules/bleno/node_modules/xpc-connection
> node-gyp rebuild

2016-04-22 15:23:04.621 xcodebuild[692:16791] [MT] PluginLoading: Required plug-in compatibility UUID ACA8656B-FEA8-4B6D-8E4A-93F4C95C362C for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/Unity4XC.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-04-22 15:23:04.623 xcodebuild[692:16791] Failed to load plugin at: /Users/abhishek/Library/Application Support/Developer/Shared/Xcode/Plug-ins/Unity4XC.xcplugin, skipping.  Reason for failure: *** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[0]
  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
  SOLINK_MODULE(target) Release/binding.node: Finished
bleno@0.4.0 node_modules/bleno
├── bplist-parser@0.0.6
├── debug@2.2.0 (ms@0.7.1)
└── xpc-connection@0.1.4 (nan@2.2.1)`

WayneKeenan, 2016-04-22:
I'd guess that your building inside a shared folder that is mounted on a Mac with a recent  (the Expected 47) version of node and that the folder is also mounted on another device (such as a Pi or other 'small' system) with a very old version of NodeJS (the Got 14)

Do:

```rm -fr node_modulde 
npm install
```

shekit, 2016-04-22:
I tried that.. still getting the same error. I also do not have this as a shared folder. It is only mounted on my mac

sandeepmistry, 2016-04-24:
@shekit please see @tdicola's post in https://github.com/sandeepmistry/noble/issues/235#issuecomment-134795340

The version of Node.js and Electron you are using probably don't use the same version of V8.

closed 2016-04-24 by sandeepmistry