noble/bleno #193: Not working on RPi3

shekit, 2016-04-23:
Hi,

I'm unable to install bleno on raspbian jessie on a pi3. I keep a bunch of errors during installation as it seems to want node version >0.12 but the current version on jessie is 0.10.29. This version seems to be the latest version and I cannot seem to upgrade it. I have node running on my path correctly as well.

WayneKeenan, 2016-04-23:
have a look at the NodeJS section of this: https://github.com/TheBubbleWorks/TheBubbleWorks_RaspberryPi_BLE_GPIO_Server/blob/master/INSTALL.md

sandeepmistry, 2016-04-24:
@shekit it's hard to help unless you share the output of the installation errors :)

shekit, 2016-04-24:
Hi @sandeepmistry I'm going to try @WayneKeenan's suggestion and report back. I think it might just be the case of updating the node version to the latest 5.x as suggested in the link

sandeepmistry, 2016-04-25:
Ok, I'll close for now then.

closed 2016-04-25 by sandeepmistry