noble/bleno #194: xpcError: connection interrupted when receiving data

jareddlc, 2016-04-25:
Hello I am trying to send data via my android app, using the android method `bluetoothGatt.writeCharacteristic(characteristic);` and on the blendo side, I get the error soon after:

`$ xpcError: connection interrupted`

what information do I need to post here to troubleshoot this issue?

OS: Mac 10.10.3
Node: v4.2.6

```
***** ble.js
bleno.on('stateChange', function(state) {
    // state = unknown | resetting | unsupported | unauthorized | poweredOff | poweredOn
    console.log('on -> stateChange: ' + state);

    if(state === 'poweredOn') {
      console.log('startAdvertising');
      bleno.startAdvertising(config.name, config.serviceUuid);
      if(callback) {
        callback(null);
      }
    }
    else {
      console.log('stopAdvertising');
      bleno.stopAdvertising();
      if(callback) {
        callback(null);
      }
    }
  });

***** characteristic.js
var BlenoCharacteristic = bleno.Characteristic;

var Characteristic = function() {
  Characteristic.super_.call(this, {
    uuid: config.characteristic.uuid,
    properties: config.characteristic.properties,
    value: config.characteristic.value,
    descriptors: [
      new bleno.Descriptor({
        uuid: config.descriptor.uuid,
        value: config.descriptor.value
      })
    ]
  });

  this._value = new Buffer(0);
  this._updateValueCallback = null;
};
util.inherits(Characteristic, BlenoCharacteristic);

***** config.js
config.name = 'App';
config.serviceUuid = ['fff0'];
config.characteristic = {
  uuid: 'fff1',
  properties: ['read', 'write'],
  value: 'App characteristic value'
};

config.descriptor = {
  uuid: 'fff2',
  value: 'App descriptor value'
};
```

jareddlc, 2016-04-26:
I also have same issue with the android app you have recommended in the README, any thoughts?

jareddlc, 2016-04-28:
I figured it out, for anyone having this problem make sure the Characteristic value is `null`

sandeepmistry, 2016-05-01:
@jareddlc this is good to know, I'll re-open this and prevent adding of characteristics with value that are non-read only.

sandeepmistry, 2016-05-02:
I've made a change to prevent characteristics with value that are non-read only in https://github.com/sandeepmistry/bleno/commit/3a1ffa9c50b0e546ca2170706c0ef0c9148f87a5.

closed 2016-05-02 by sandeepmistry

jareddlc, 2016-05-02:
👍