noble/bleno #197: How fast can data be sent

rpendleton, 2016-05-03:
In iOS, peripherals have the ability to send notifications very quickly until the queue is full. Once it's full, the notify method returns false, indicating that the peripheral must stop sending data. Once the queue has more space, the app is notified via a delegate call and can resume sending data. This provides an easy means of sending large amounts of data in a stream without any loss or reordering of data.

The corresponding sample code can be found in Apple's [documentation][1]. The delegate call is also [documented][2]. Is there any way to do something similar with bleno? I'm currently in need of a way to stream data back and forth between two devices. With bleno, when I write too much data, hci0 becomes completely unresponsive and I have to restart the Pi in order to fix it. This is similar to #146 and #179.

If there isn't a way to determine if data can be sent, is there a way to determine how quickly data may be sent in general? I believe this has something to do with the number of packets sent during the connection interval, but I've been unable to find a reliable method for determining this.

[1]: https://developer.apple.com/library/ios/samplecode/BTLE_Transfer/Introduction/Intro.html#//apple_ref/doc/uid/DTS40012927-Intro-DontLinkElementID_2

[2]: https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBPeripheralManagerDelegate_Protocol/#//apple_ref/occ/intfm/CBPeripheralManagerDelegate/peripheralManagerIsReadyToUpdateSubscribers:

sandeepmistry, 2016-05-09:
See answer in https://github.com/sandeepmistry/bleno/issues/179#issuecomment-217754032.

Also, maybe this can already be done via the `onNotify` and `onIndicate` callback hooks. On OS X, the CoreBluetooth hooks you mention are already use for the callback hooks I mentioned.

sandeepmistry, 2016-05-16:
I'm closing this for now, I think the question has been answered and as mentioned in https://github.com/sandeepmistry/bleno/issues/179#issuecomment-218763159 seems to work.

closed 2016-05-16 by sandeepmistry