noble/bleno #198: Can't fetch characteristic / LightBlue crashes while running on Raspberry Pi and iOS

methyl, 2016-05-18:
When I run bleno peripherial on RPi, eg. the `echo` example or `test.js` script and open LightBlue on OSX, everything runs properly. When I switch to iOS's LightBlue, the program crashes while connecting. In simple iOS application I wrote (also using `CoreBluetooth`), result is also not appealing - it doesn't detect any characteristic while running bleno on Raspberry. If I run bleno on OSX, once again everything is in working order. Both iOS LightBlue and my app are good. 

It seems there is some problem with Raspberry PI host and iOS guest, but I have no idea how should I debug any further. 

I use newest Raspbian - jessie, Node 6.1 and BlueZ 5.23. `bluetoothd` is stopped and don't appear in process list.

methyl, 2016-05-18:
I attach hcidump for following flow:

1. `node examples/echo/main.js`
2. connect using my iOS application
3. restart `node examples/echo/main.js`
4. connect using iOS LightBlue

[hcidump.txt](https://github.com/sandeepmistry/bleno/files/270003/hcidump.txt)

sandeepmistry, 2016-06-19:
@methyl 

I'm seeing the Raspberry Pi send a disconnect ~20s after connecting:

```
2016-05-18 08:39:48.365980 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role slave
      bdaddr 5F:6B:88:4E:50:E0 (Random)

... 

2016-05-18 08:40:08.030879 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection

...

2016-05-18 08:40:14.633294 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role slave
      bdaddr 5F:6B:88:4E:50:E0 (Random)

```

What version of `node-bluetooth-hci-socket` do you have installed?

methyl, 2016-06-20:
Hey @sandeepmistry, the problem disappeared after restarting iPhone, not sure what was the problem but it seems it won't appear again. Thanks for taking a look!

closed 2016-06-20 by methyl

Stan92, 2017-01-19:
Hi @methyl 
I do confirm, the problem disappears after restarting the iOS device.
But the question is why this happens :-) did you find a way to prevent this behavior?

methyl, 2017-01-20:
It was just one time it happened to me, but I didn't use bleno extensively. Wonder if problem is with bleno or iOS itself.