noble/bleno #199: Impossible to accept Write command from nRF51822 chipset at RPi2 with Bleno running

vovojh, 2016-05-19:
I plan to make RPi2 function as a peripheral role with a customary service.

The service should maintain one 'Write' characteristic and nRF51822 chipset sends ATT command to change(write into) the value of the characteristic.

I edited "bleno/test.js" to rule out all the characteristics except one having write property and changed the service UUID and the characteristic UUID according to the sample "central" program for nRF51822.

The following is about the sample code used for nRF 51822.
https://devzone.nordicsemi.com/tutorials/21/

It seems like connection is somehow established between them, but write request is not handled at RPi2.

I also tested LightBlue app (in iPhone)  and it worked with no problem.

I ran "hcidump -t -X" and outputs are as follows.

[withiPhone.txt](https://github.com/sandeepmistry/bleno/files/272685/withiPhone.txt)
[withnRF.txt](https://github.com/sandeepmistry/bleno/files/272686/withnRF.txt)

The kernel version is as follows.
[image]
[image]

I installed and tested bluez-5.39, 4.99, 4.101.

All of them showed similar results. 

[image]

In case of connection with iPhone, response to write request showed up.

I also ran rpi-update to downgrade kernel version to 3.12.36+ according to following link.
http://stackoverflow.com/questions/34966858/connecting-ble-device-to-raspberry-fails

This is how i installed bluez. ( i installed D-BUS, and any of dependent packages via apt-get)

./configure --disable-systemd --enable-library
sudo make
sudo make install

I cannot see the write request from bleno.js side but i can see write request is coming from bluez side.
(both hcidump & btmon showed write ATT command had wrong handle value)

I suspect ATT information exchange (?) failed with nRF51822 chipset, however succeeded with iPhone.

I also read many related issues here and tried 'sudo service stop bluetooth'.
It showed no bluetoothd is launched? and i thought no GATT server is running at bluez side.
so i concluded there is no GATT server running to intercept ATT command message.(which im not so sure about)

I don't know what to do and where to start.

I keep eyes on this issue and can get back to any advice ASAP.

Please let me know any more information is required regarding this issue.

Thanks!

vovojh, 2016-05-19:
i forgot to mention this. LightBlue App can copy the service from RPi2 and mimic it. nRf 51822 succeeded to send write command to LightBlue ios app.

sandeepmistry, 2016-06-19:
@vovojh it looks like your nRF51 app is trying to write to handle 0x0000:

```
    ATT: Write cmd (0x52)
      handle 0x0000 value  0x31 0x0a
```

Which is always an invalid handle. I'm closing this for now, doesn't seem to be anything related to bleno.

closed 2016-06-19 by sandeepmistry