noble/bleno #201: crash on ubuntu

farfromrefug, 2016-06-04:
I am running bleno on a ubuntu machine which does not have a compatible BLE adapter.
I know bleno should not work but it should not crash ;)
```
bleno warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] BLENO_HCI_DEVICE_ID=x node ...
on -> stateChange: unsupported true
RangeError: index out of range
    at checkOffset (buffer.js:663:11)
    at Buffer.readUInt8 (buffer.js:701:5)
    at [object Object].Hci.processCmdCompleteEvent (/home/mguillon/dev/akylas.aryballe.neoseserver/node_modules/bleno/lib/hci-socket/hci.js:503:21)
    at [object Object].Hci.onSocketData (/home/mguillon/dev/akylas.aryballe.neoseserver/node_modules/bleno/lib/hci-socket/hci.js:427:12)
    at emitOne (events.js:90:13)
    at BluetoothHciSocket.emit (events.js:182:7)

```

sandeepmistry, 2016-06-06:
@farfromrefug I've corrected this in https://github.com/sandeepmistry/bleno/commit/d49e76bcc7fa248d656c3635adcec65a83f4b029.

You can try it out before the next publish to npm via: `npm install sandeepmistry/bleno`

closed 2016-06-06 by sandeepmistry

farfromrefug, 2016-06-06:
@sandeepmistry thanks a lot. I can confirm that it works great! Though strangely i don't see the warning about BLE anymore (might be unrelated).
Thanks again

sandeepmistry, 2016-06-19:
@farfromrefug cool, thanks for verifying my change :)