noble/bleno #203: Fedora 24, BlueZ 5.40 - Can't connect

ecdpalma, 2016-06-28:
I've seen some issues describing a similar behaviour, but I'm opening a new issue for the specifics.

Running example code, the peripheral advertisement is received by the master device (an Android with the Nordic app), but the "discovering services" operation fails with error 133 or 129 (as logged by the app). I tested previously with Fedora 23 with the same results.

sandeepmistry, 2016-07-03:
Hi @ecdpalma,

As per #24 and the read me (https://github.com/sandeepmistry/bleno#linux). You'll have to stop `bluetoothd` when using the current version of bleno.

ecdpalma, 2016-07-03:
Hi @sandeepmistry,

I've done that. Checkeg if there were some process, but can't connect anyway. I'll try to debug it more during the week and update here.

sandeepmistry, 2016-07-04:
@ecdpalma ok. Please also supply the following:

* Debug logs: `sudo DEBUG=* node <file>.js`
* HCI dump capture: `sudo hcidump -t -x`

ecdpalma, 2016-07-10:
Hi @sandeepmistry,

Solved. I've been able to get it working after I looked closely into the hcidump output. Misteriously, even with bluetoothd stopped, scan in Fedora was kept or being enabled. I had to disable bluetoothd and also rfkill block and then unblock. Thus, my steps are:

```
systemctl stop bluetooth.service
systemctl disable bluetooth.service
rfkill block bluetooth
rfkill unblock bluetooth
hciconfig hci0 up
```

And then run the js code.

closed 2016-07-10 by ecdpalma

sandeepmistry, 2016-07-13:
Hi @ecdpalma, thanks for the update.

Do you think it would be useful to others to add this to the read me? https://github.com/sandeepmistry/noble#fedora--other-rpm-based

If yes, could you please submit a pull request for this.