noble/bleno #205: No BLE discovered by nRF Master

bzhong2, 2016-07-09:
A few days ago, when I ran pizza example, every thing works. However, after the new windows 10 update (probably because of the update), when I ran node peripheral, it shows start advertising... but nRF master  (android) can not discover the BLE. Has any one faced with a similar problem?

bzhong2, 2016-07-11:
I reinstalled the system to be windows 8.1 and it works well at the first time. After I restart the computer, the same problem appears again. It shows "start advertising" but nRF master (android) cannot find this BLE anymore. I tried to turn off my firewall and virus protection but no help.

sandeepmistry, 2016-07-13:
@bzhong2 have you tried turning Bluetooth off and on or rebooting your Android device?

Could you provide some debug logs, by setting the DEBUG environment variable to `*`?

bzhong2, 2016-07-13:
I ran "set DEBUG =*, -not_this". And I ran the code and got the debug logs as below. I tried turing Bluetooth off and on or reboot Android. I also tried reinstall the driver of usb bluetooth Adapter. Thanks a lot for your help! 
E:\Install\Nodejs\bleno-master\examples\pizza>node peripheral
  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('5a4258', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +0ms
  bleno platform win32 +101ms
  hci set advertise enable - writing: 010a200100 +7ms
  hci-usb write: 010a200100 +4ms
E:\Install\Nodejs\bleno-master\node_modules\usb\usb.js:33
        this.__open()
             ^

Error: LIBUSB_ERROR_ACCESS
    at Error (native)
    at Device.usb.Device.open (E:\Install\Nodejs\bleno-master\node_modules\usb\usb.js:33:7)
    at BluetoothHciSocket.bindUser (E:\Install\Nodejs\bleno-master\node_modules\bluetooth-hci-socket\lib\usb.js:53:19)
    at BluetoothHciSocket.bindRaw (E:\Install\Nodejs\bleno-master\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (E:\Install\Nodejs\bleno-master\lib\hci-socket\hci.js:100:18)
    at BlenoBindings.init (E:\Install\Nodejs\bleno-master\lib\hci-socket\bindings.js:97:13)
    at new Bleno (E:\Install\Nodejs\bleno-master\lib\bleno.js:46:18)
    at Object.<anonymous> (E:\Install\Nodejs\bleno-master\index.js:3:18)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)

bzhong2, 2016-07-20:
When I tried to use BLE usb doggle instead of laptop in-built BLE adapter. This problem has been solved by now. Thanks a lot!

closed 2016-07-20 by bzhong2