noble/bleno #206: Can't run on multiple adapters

mhoresh, 2016-07-11:
I tried to run two instances of node, each of which implementing the same service, one on hci0 and the other on hci1 (using the BLENO_HCI_DEVICE_ID environment variable) . Only one of the services worked properly. the other one stopped advertising.

rodrigosalinas, 2016-07-17:
I have the very same problem in Ubuntu 15.04 and Node 6.

sandeepmistry, 2016-07-18:
@mhoresh @rodrigosalinas please provide debug lots:

```
sudo DEBUG=* BLENO_HCI_DEVICE_ID=0 <file>.js
sudo DEBUG=* BLENO_HCI_DEVICE_ID=1 <file>.js
```

Also, have you stopped bluetoothd if you are using BlueZ 5.x?

mhoresh, 2016-07-24:
I am using BlueZ 5.x and have stopped the bluetooth service. 

This is the log from HCI1:
  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('77732d6d6f727269732d7562756e7475', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +0ms
  bleno platform linux +5ms
  hci setting filter to: 1400000020c10000000000400000 +3ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +0ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +0ms
[... 736 lines cut ...]
  hci 		result =  +0ms
  mgmt on data ->0b0001000d00dc6c0088867b02000000000000 +1s
  hci onSocketData: 040e04010b2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e04010c2000 +2ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  mgmt on data ->0c0001000800dc6c0088867b0203 +6s
  hci onSocketData: 040e04010c2000 +34s
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  mgmt on data ->050001000000 +11ms
  mgmt on data ->040001000000 +735ms

HCI1 stopped advertising.

TomThats, 2016-08-07:
I have the same problem on a raspberry pi 3 running two USB Bluetooth Sticks.

Both have their own device assigned via BLENO_HCI_DEVICE_ID.

If I run the scripts with all debug info, I can see that both scripts receive the same data from the sockets even though just one is used with an app.

Script #1:
```
  hci set advertise enable - writing: 010a200101 +1ms
  mgmt on data ->0c0000000800c44b8bf78a440203 +0ms
  hci onSocketData: 040e04010a2000 +54ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8202 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
```

Script #2:
```
 bleno servicesSet +5ms
  hci onSocketData: 040e0401092000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 8201 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0401082000 +0ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8200 +0ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  mgmt on data ->0b0000000d00c44b8bf78a4402000000000000 +13s
  mgmt on data ->0c0000000800c44b8bf78a440203 +3s
```

I'm not a expert but that shouldn't happen if both scripts use their own device, or am I wrong?

sandeepmistry, 2016-09-07:
Hi @mhoresh, @rodrigosalinas, @TomThats,

Could you please try the latest master version `npm install sandeepmistry/bleno`. It uses the latest `bluetooth-hci-socket` version that includes the following fix: https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/14dc3660f9e86695c6c75b92dc2222da2905549e

Let me know if it works, then I can publish a new version to npm.

closed 2016-09-07 by sandeepmistry

mhoresh, 2016-09-07:
It seems to be working well for me. Thank you.