noble/bleno #207: automatically accept client b8:27:eb:00:82:2d

AlessandroFerri, 2016-07-14:
Hi, 
is happening strange thing.
I have two raspberry, in the first I have a complete project developed with blend, for other things, and a Bluetooth address b8:27:eb:00:82:2d, in the second I'm developing a new project. I have the first raspberry on and a service bleno that running, when I run the blend service test in the second raspberry, this automatically accepts the connection from the first raspberry client,
```
pi@raspberrypi:~/New_Project/iBeacon $ sudo node test
blend
on -> stateChange: poweredOn, address = b8:27:eb:af:61:7f
on -> advertisingStart: success
on -> servicesSet: success
on -> accept, client: b8:27:eb:00:82:2d
on -> rssiUpdate: -63
```
preventing me to connect from iPad. I don't no why, How can I avoid this?

sandeepmistry, 2016-07-18:
Hi @AlessandroFerri,

There's nothing in bleno to auto connect to another device. Maybe you've configured something else in the other Raspberry Pi to do so?

You can call `bleno.disconnect()` to manually disconnect it.

closed 2016-07-18 by sandeepmistry

AlessandroFerri, 2016-07-18:
Hi and thank for reply,
the only thing present in the other raspberry is always `bleno peripherals`. In this raspberry I installed bleno to configure a a2dp service from my iPad to listen music while in my new raspberry, I'm installing bleno to configure an ADC (Analog to Digital Converter). Once, I simply paired the two bluetooth services of the raspberry, to send the music from the new to the old raspberry, but since then, every time, even if the two raspberry are not paired, the `bleno peripherals` in the new raspberry automatically connects to the old raspberry `bleno peripherals`, while I I want to connect both with my `central` App on my iPad.
I have tried removing the device with `bluetoothctl`, but nothing changes.

AlessandroFerri, 2016-07-18:
Hi,
Excuse me if I insist but I do not understand why my new raspberry auto accepts client. I did a test by putting the same copy of SSD image of the old raspberry on another raspberry, changing the BT ID and the issue does not occur. I send the DEBUG, hoping that you have some help.

```
pi@raspberrypi:~/node_modules/bleno $ sudo DEBUG=* node test
  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('7261737062657272797069', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +0ms
  bleno platform linux +23ms
  hci setting filter to: 1400000020c10000000000400000 +14ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +1ms
[... 288 lines cut ...]
  hci 			master clock accuracy = 1 +0ms
  bleno accept b8:27:eb:00:82:2d +4ms
on -> accept, client: b8:27:eb:00:82:2d
  hci read rssi - writing: 010514024000 +1ms
  mgmt on data ->0b0000000d002d8200eb27b801000000000000 +1ms
  hci onSocketData: 040f0400011620 +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 15 +0ms
  hci onSocketData: 043e0c041a40001f00000000000000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 4 +5ms
  hci 		LE meta event status = 26 +0ms
  hci 		LE meta event data = 40001f00000000000000 +0ms
  hci onSocketData: 040e07010514004000c0 +3ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +5ms
  hci 		cmd = 5125 +0ms
  hci 		status = 0 +0ms
  hci 		result = 4000c0 +0ms
  hci 			handle = 64 +0ms
  hci 			rssi = -64 +0ms
on -> rssiUpdate: -64

```