noble/bleno #210: Automatic Accept client, without request

AlessandroFerri, 2016-07-20:
Hi,
I had opened the issue # 207 on GitHub, but you've closed. I need your help because I can not understand the problem.
if I manually disconnect with `bleno.disconnect()`, it automatically reconnects and code enters in a loop. There must be something else to prevent this automatic connection.
Excuse me if I insist but I do not understand why my new raspberry auto accepts client. I did a test by putting the same copy of SSD image of the old raspberry on another raspberry, changing the BT ID and the issue does not occur.

**Is there a list of ID-address somewhere that makes a Bluetooth BLE reconnect when the device is nearby?**

I send the DEBUG, hoping that you have some help.

```
pi@raspberrypi:~/node_modules/bleno $ sudo DEBUG=* node test
  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('7261737062657272797069', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
[... 293 lines cut ...]
  hci           supervision timeout = 420 +0ms
  hci           master clock accuracy = 1 +0ms
  bleno accept b8:27:eb:00:82:2d +4ms
on -> accept, client: b8:27:eb:00:82:2d
  hci read rssi - writing: 010514024000 +1ms
  mgmt on data ->0b0000000d002d8200eb27b801000000000000 +1ms
  hci onSocketData: 040f0400011620 +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 15 +0ms
  hci onSocketData: 043e0c041a40001f00000000000000 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +0ms
  hci       LE meta event type = 4 +5ms
  hci       LE meta event status = 26 +0ms
  hci       LE meta event data = 40001f00000000000000 +0ms
  hci onSocketData: 040e07010514004000c0 +3ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +5ms
  hci       cmd = 5125 +0ms
  hci       status = 0 +0ms
  hci       result = 4000c0 +0ms
  hci           handle = 64 +0ms
  hci           rssi = -64 +0ms
on -> rssiUpdate: -64
```

AlessandroFerri, 2016-07-21:
I have noticed that the only differences between the old and the new respberry is:

for the old:

```
pi@Ani ~ $ sudo hciconfig hci0 lestates
Supported link layer states:
	YES Non-connectable Advertising State
	YES Scannable Advertising State
	YES Connectable Advertising State
	YES Directed Advertising State
	YES Passive Scanning State
	YES Active Scanning State
	YES Initiating State/Connection State in Master Role
	YES Connection State in the Slave Role
	YES Non-connectable Advertising State and Passive Scanning State combination
	YES Scannable Advertising State and Passive Scanning State combination
	YES Connectable Advertising State and Passive Scanning State combination
	YES Directed Advertising State and Passive Scanning State combination
	YES Non-connectable Advertising State and Active Scanning State combination
	YES Scannable Advertising State and Active Scanning State combination
	YES Connectable Advertising State and Active Scanning State combination
	YES Directed Advertising State and Active Scanning State combination
	YES Non-connectable Advertising State and Initiating State combination
	YES Scannable Advertising State and Initiating State combination
	YES Non-connectable Advertising State and Master Role combination
	YES Scannable Advertising State and Master Role combination
	YES Non-connectable Advertising State and Slave Role combination
	YES Scannable Advertising State and Slave Role combination
	NO  Passive Scanning State and Initiating State combination
	NO  Active Scanning State and Initiating State combination
	YES Passive Scanning State and Master Role combination
	YES Active Scanning State and Master Role combination
	YES Passive Scanning State and Slave Role combination
	YES Active Scanning State and Slave Role combination
	YES Initiating State and Master Role combination/Master Role and Master Role combination
```
for the new:
```
pi@Ani ~ $ sudo hciconfig hci0 lestates
Supported link layer states:
	YES Non-connectable Advertising State
	YES Scannable Advertising State
	YES Connectable Advertising State
	YES Directed Advertising State
	YES Passive Scanning State
	YES Active Scanning State
	YES Initiating State/Connection State in Master Role
	YES Connection State in the Slave Role
	YES Non-connectable Advertising State and Passive Scanning State combination
	YES Scannable Advertising State and Passive Scanning State combination
	YES Connectable Advertising State and Passive Scanning State combination
	YES Directed Advertising State and Passive Scanning State combination
	YES Non-connectable Advertising State and Active Scanning State combination
	YES Scannable Advertising State and Active Scanning State combination
	YES Connectable Advertising State and Active Scanning State combination
	YES Directed Advertising State and Active Scanning State combination
	YES Non-connectable Advertising State and Initiating State combination
	YES Scannable Advertising State and Initiating State combination
	YES Non-connectable Advertising State and Master Role combination
	YES Scannable Advertising State and Master Role combination
	YES Non-connectable Advertising State and Slave Role combination
	YES Scannable Advertising State and Slave Role combination
	YES Passive Scanning State and Initiating State combination
	YES Active Scanning State and Initiating State combination
	YES Passive Scanning State and Master Role combination
	YES Active Scanning State and Master Role combination
	YES Passive Scanning State and Slave Role combination
	YES Active Scanning State and Slave Role combination
	YES Initiating State and Master Role combination/Master Role and Master Role combination
```
ie the following two parameters,
```
Passive Scanning State and Initiating State combination
Active Scanning State and Initiating State combination
```
are set to YES instead of NO, are these that make the difference and how I can change?

AlessandroFerri, 2016-07-27:
Please, ask someone help!!

AlessandroFerri, 2016-07-27:
Among my various tests and trials I have found something that perhaps can be indicative.
My ID of the raspberry is `FF:FF:FF:FF:FF:F0`, if  I delete the folder ID of hci0 as follows:
```
root@Ani:~# cd /var/lib/bluetooth
root@Ani:/var/lib/bluetooth# ls
FF:FF:FF:FF:FF:F0
root@Ani:/var/lib/bluetooth# rm -R FF:FF:FF:FF:FF:F0
```
and I reboot the raspberry, the issue still remains,
```
pi@Ani ~/New_Project/iBeacon $ sudo node New_Rev2
BleAni-fffff0

bleno
on -> stateChange: poweredOn, address = undefined
on -> advertisingStart: success
on -> servicesSet: success
on -> accept, client: B8:27:EB:00:82:2D
on -> rssiUpdate: 127
^Cpi@Ani ~/New_Project/iBeacon $ 
```
but if I change the hci0 ID, for example from `FF:FF:FF:FF:FF:F0` to `FF:FF:FF:FF:FF:F1`, as follows
`sudo bdaddr -i hci0 -r FF:FF:FF:FF:FF:F1`
the problem no longer exists,
```
pi@Ani ~/New_Project/iBeacon $ sudo node New_Rev2
BleAni-fffff1

bleno
on -> stateChange: poweredOn, address = undefined
on -> advertisingStart: success
on -> servicesSet: success
^Cpi@Ani ~/New_Project/iBeacon $ 
```
also if I have the old dir of the precedent ID,
```
root@Ani:~# cd /var/lib/bluetooth
root@Ani:/var/lib/bluetooth# ls
FF:FF:FF:FF:FF:F0  FF:FF:FF:FF:FF:F1
root@Ani:/var/lib/bluetooth# 
```
It seems that the ID of the old raspberry (`B8:27:EB:00:82:2D`) is recorded **somewhere else**, but I do not understand where.
I hope this gives some ideas.

AlessandroFerri, 2016-08-01:
I want to add more information. When I connect my iPad to my old raspberry, the connection type is:
```
pi@Ani ~ $ hcitool con
Connections:
	> LE 7A:35:74:57:DD:A8 handle 67 state 1 lm SLAVE 
```
When I start the service bleno from my new raspberry, this automatically connects to the bleno service of my old raspberry and (from my old raspberry)the connection type is:
```
pi@Ani ~ $ hcitool con
Connections:
	< LE B8:27:EB:D2:06:6C handle 71 state 1 lm MASTER 

```
but from my new raspberry the connection type is:
```
pi@Ani ~ $ hcitool con
Connections:
	< Unknow B8:27:EB:00:82:2D handle 71 state 1 lm SLAVE 

```
So in the old raspberry changes from `SLAVE` to `MASTER` type of connection

AlessandroFerri, 2016-08-17:
After various tests, it seems that between the two raspberry (RPi2 and RPI3), it is not necessary to run the service bleno on both for auto accept RPi2 client. After pared, but not trusted, the RPi2 from RPi3, if stop the bleno service on RPi2 and I run the same on RPi3, bleno auto accept the RPi2 client and I see on RPi2:
```
pi@raspberriPI2 ~$ hcitool con
Connections:
	< LE B8:27:EB:D2:06:6C handle 39 state 1 lm MASTER
```
and on RPi3:
```
pi@raspberryPI3 ~.../iBeacon $ sudo node New_Rev2
bleno
on -> stateChange: poweredOn, address = B8:27:EB:D2:06:6C
on -> advertisingStart: success
on -> servicesSet: success
on -> accept, client: B8:27:EB:00:82:2D
on -> rssiUpdate: 127
```

AlessandroFerri, 2016-08-24:
**I have found the solution myself.**
In solving the issue, I have also solved the issue to disable bluetoothd, for Ubuntu/Debian/Raspbian, if BlueZ 5.14 or later is installed, indicated as a prerequisite in the README.
Now you no longer need, Bluetooth can also be used for different connections such as A2DP or other without interfering with bleno.
If anyone is interested, just leave a request on this page (issue # 210) and I will send, as a comment, the procedure that I have done.

sandeepmistry, 2016-09-05:
Hi @AlessandroFerri,

I'm glad to hear you found a solution.

For future reference, contacting me via multiple avenues (Github issues, Gitter, Twitter, direct email) multiple times is not a great way to get my attention (it actually has the opposite effect).

closed 2016-09-05 by sandeepmistry

kjartanhefring, 2020-12-09:
> **I have found the solution myself.**
> In solving the issue, I have also solved the issue to disable bluetoothd, for Ubuntu/Debian/Raspbian, if BlueZ 5.14 or later is installed, indicated as a [...]

Hello @AlessandroFerri, could you share what you have done to fix this? I am dealing with a similar problem. Thank you.