noble/bleno #213: Pairing error Unspecified Reason

adi1133, 2016-08-06:
I run linux, here is my log file

> 2016-08-06 19:04:14.325726 > ACL data: handle 64 flags 0x02 dlen 11
>     SMP: Pairing Request (0x01)
>       capability 0x04 oob 0x00 auth req 0x0d
>       max key size 0x10 init key dist 0x07 resp key dist 0x07
>       Capability: KeyboardDisplay (OOB data not present)
>       Authentication: Bonding (MITM Protection)
>       Initiator Key Distribution:  LTK IRK CSRK
>       Responder Key Distribution:  LTK IRK CSRK
> 2016-08-06 19:04:14.327429 < ACL data: handle 64 flags 0x00 dlen 11
>     SMP: Pairing Response (0x02)
>       capability 0x03 oob 0x00 auth req 0x01
>       max key size 0x10 init key dist 0x00 resp key dist 0x01
>       Capability: NoInputNoOutput (OOB data not present)
>       Authentication: Bonding (No MITM Protection)
>       Initiator Key Distribution:    
>       Responder Key Distribution:  LTK  
> 2016-08-06 19:04:14.460943 > ACL data: handle 64 flags 0x02 dlen 21
>     SMP: Pairing Confirm (0x03)
>       key 9d3cf4e56c7baaad41633a960ff35ff8
> 2016-08-06 19:04:14.472950 < ACL data: handle 64 flags 0x00 dlen 21
>     SMP: Pairing Confirm (0x03)
>       key f9396cb524e03b0488ac19190b62cf1f
> 2016-08-06 19:04:14.589850 > HCI Event: Number of Completed Packets (0x13) plen 5
>     handle 64 packets 1
> 2016-08-06 19:04:14.595937 > ACL data: handle 64 flags 0x02 dlen 21
>     SMP: Pairing Random (0x04)
>       random 7c50d96f94ab0498d03aa9805d08c3c2
> 2016-08-06 19:04:14.598447 < ACL data: handle 64 flags 0x00 dlen 21
>     SMP: Pairing Random (0x04)
>       random 6fe489c64aca892d5a46e1c0d5dd0af9
> 2016-08-06 19:04:14.730772 > HCI Event: Number of Completed Packets (0x13) plen 5
>     handle 64 packets 2
> 2016-08-06 19:04:14.806343 > HCI Event: LE Meta Event (0x3e) plen 10
>     LE Connection Update Complete
>       status 0x00 handle 64
>       interval 7.50ms, latency 0.00ms, superv. timeout 20000.00ms
> 2016-08-06 19:04:14.806597 > HCI Event: LE Meta Event (0x3e) plen 13
>     LE Long Term Key Request
>     40 00 00 00 00 00 00 00 00 00 00 00 
> 2016-08-06 19:04:14.806646 < HCI Command: LE Long Term Key Request Negative Reply (0x08|0x001b) plen 2
>   40 00 
> 2016-08-06 19:04:14.807117 > HCI Event: Command Complete (0x0e) plen 6
>     LE Long Term Key Request Negative Reply (0x08|0x001b) ncmd 1
>     00 40 00 
> 2016-08-06 19:04:14.807682 < ACL data: handle 64 flags 0x00 dlen 6
>     SMP: Pairing Failed (0x05)
>       reason 0x08
>       Reason Unspecified Reason
> 2016-08-06 19:04:14.933685 > HCI Event: LE Meta Event (0x3e) plen 10
>     LE Connection Update Complete
>       status 0x00 handle 64
>       interval 67.50ms, latency 0.00ms, superv. timeout 20000.00ms
> 2016-08-06 19:04:15.089876 > HCI Event: Number of Completed Packets (0x13) plen 5
>     handle 64 packets 1
> 
>

adi1133, 2016-08-06:
I forgot to disable bluetoothd, not I get the same error as #208 
Couldn't pair with Battery because of an incorrect PIN or passkey.

adi1133, 2016-08-06:
Somehow managed to pair the device (after a reboot).
Now after the connection happens I get status 0x13 and a disconnect.

> 2016-08-06 20:49:06.483951 > HCI Event: Command Status (0x0f) plen 4
>     LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> 2016-08-06 20:49:06.484927 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Enable (0x08|0x000c) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.485517 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Parameters (0x08|0x000b) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.486025 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Enable (0x08|0x000c) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.489330 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Enable (0x08|0x000c) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.489949 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Parameters (0x08|0x000b) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.490525 > HCI Event: Command Complete (0x0e) plen 4
>     LE Set Scan Enable (0x08|0x000c) ncmd 1
>     status 0x00
> 2016-08-06 20:49:06.742134 > HCI Event: Command Complete (0x0e) plen 6
>     LE Long Term Key Request Reply (0x08|0x001a) ncmd 1
>     00 40 00 
> 2016-08-06 20:49:06.936216 > HCI Event: Encrypt Change (0x08) plen 4
>     status 0x00 handle 64 encrypt 0x01
> 2016-08-06 20:49:10.747493 > HCI Event: Command Status (0x0f) plen 4
>     Disconnect (0x01|0x0006) status 0x00 ncmd 1
> 2016-08-06 20:49:10.863063 > HCI Event: Disconn Complete (0x05) plen 4
>     status 0x00 handle 64 reason 0x16
>     Reason: Connection Terminated by Local Host
>

sandeepmistry, 2016-08-21:
@adi1133 which kernel version are you using?

sandeepmistry, 2016-09-11:
I'm closing this do to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-09-11 by sandeepmistry

adi1133, 2016-09-11:
I am interested but my project is postponed. I will post more info when available.