noble/bleno #214: Characteristic.onUnsubscribe not called on disconnect

adi1133, 2016-08-10:
When a blue tooth device gets disconnected, its notify listeners remain active.

sandeepmistry, 2016-10-02:
Hi @adi1133, @jor3l, @henla464,

This has been implemented in https://github.com/sandeepmistry/bleno/commit/1a3cbd49bbaf7d024faa4e0a8d3bc4d17d25cc99. Please try it our via `npm install sandeepmistry/bleno` and provide your feedback.

closed 2016-10-02 by sandeepmistry

mattsson, 2017-02-23:
We're having the same issue. When will this be released?

sandeepmistry, 2017-03-13:
@mattsson I've published v0.4.2 to npm now.

mattsson, 2017-03-13:
@sandeepmistry Cool, thanks.

erezc74, 2017-09-24:
Hi @sandeepmistry, 
Was this issue fixed in 0.4.2?
I'm using 0.4.2 and still onUnsubscribe not being called on disconnect.
I'm wondering if there is an issue on my end or maybe it wasn't fixed yet?
10x.

erezc74, 2017-09-24:
Hi @sandeepmistry ,
Fixed it by adding the code in [1a3cbd4](https://github.com/sandeepmistry/bleno/commit/1a3cbd49bbaf7d024faa4e0a8d3bc4d17d25cc99) manually.
10x