noble/bleno #217: crash during the bonding

leducb, 2016-08-18:
Hi sandeepmistry and thank for your work !

I am here because I am facing an unexpected problem when i'm trying to bonding my phone (android 6.0.1/ OP3) to my rasp on ubutu mate (LTS) with the included bluetooth adapter.

this is the crash log :  

`/home/****/node_modules/bleno/lib/hci-socket/crypto.js:60
  var output = new Buffer(input.length);
                               ^

TypeError: Cannot read property 'length' of undefined
    at swap (/home/****u/node_modules/bleno/lib/hci-socket/crypto.js:60:32)
    at e (/home/****/node_modules/bleno/lib/hci-socket/crypto.js:37:9)
    at Object.c1 (/home/****/node_modules/bleno/lib/hci-socket/crypto.js:22:9)
    at Smp.handlePairingRandom (/home/****/node_modules/bleno/lib/hci-socket/smp.js:130:12)
    at Smp.onAclStreamData (/home/****/node_modules/bleno/lib/hci-socket/smp.js:58:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/home/****/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/home/****/node_modules/bleno/lib/hci-socket/bindings.js:198:21)
    at emitThree (events.js:97:13)`

leducb, 2016-08-18:
the hcidump output : 

 bleno servicesSet +3ms
  hci onSocketData: 040e0401092000 +2ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 8201 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0401082000 +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8200 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 043e13010044000101a5e4e96f514f24000000d00700 +8s
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 1 +1ms
  hci 		LE meta event status = 0 +0ms
  hci 		LE meta event data = 44000101a5e4e96f514f24000000d00700 +0ms
  hci 			handle = 68 +2ms
  hci 			role = 1 +1ms
  hci 			address type = random +0ms
  hci 			address = 4f:51:6f:e9:e4:a5 +0ms
  hci 			interval = 45 +0ms
  hci 			latency = 0 +0ms
  hci 			supervision timeout = 20000 +1ms
  hci 			master clock accuracy = 0 +0ms
  bleno accept 4f:51:6f:e9:e4:a5 +13ms
  mgmt on data ->0b0001000d00a5e4e96f514f02000000000000 +1ms
  hci onSocketData: 02440010000c00050012010800280038000000d007 +4ms
  hci 	event type = 2 +1ms
  hci onSocketData: 0244200a0006000500130102000000 +44ms
  hci 	event type = 2 +1ms
  hci 		cid = 5 +1ms
  hci 		handle = 68 +0ms
  hci 		data = 130102000000 +0ms
  hci onSocketData: 0244200b0007000400100100ffff0028 +2ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 68 +0ms
  hci 		data = 100100ffff0028 +0ms
  gatt handing request: 100100ffff0028 +30ms
  gatt read by group: startHandle = 0x1, endHandle = 0xffff, uuid = 0x2800 +8ms
  gatt response: 1106010005000018060009000118 +8ms
  gatt send: 1106010005000018060009000118 +2ms
  hci write acl data pkt - writing: 02440012000e0004001106010005000018060009000118 +3ms
  hci onSocketData: 0244200b0007000400100a00ffff0028 +82ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 68 +1ms
  hci 		data = 100a00ffff0028 +0ms
  gatt handing request: 100a00ffff0028 +0ms
  gatt read by group: startHandle = 0xa, endHandle = 0xffff, uuid = 0x2800 +1ms
  gatt response: 11140a000f001bc5d5a50200c2a1e4111809c0d6daae +1ms
  gatt send: 11140a000f001bc5d5a50200c2a1e4111809c0d6daae +0ms
  hci write acl data pkt - writing: 0244001a001600040011140a000f001bc5d5a50200c2a1e4111809c0d6daae +1ms
  hci onSocketData: 0244200b0007000400101000ffff0028 +83ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +1ms
  hci 		handle = 68 +0ms
  hci 		data = 101000ffff0028 +1ms
  gatt handing request: 101000ffff0028 +0ms
  gatt read by group: startHandle = 0x10, endHandle = 0xffff, uuid = 0x2800 +1ms
  gatt response: 011010000a +1ms
  gatt send: 011010000a +1ms
  hci write acl data pkt - writing: 024400090005000400011010000a +0ms
  hci onSocketData: 0244200b000700040008010005000228 +84ms
  hci 	event type = 2 +0ms
[... 157 lines cut ...]
  hci 		cid = 6 +0ms
  hci 		handle = 68 +1ms
  hci 		data = 0104000d100707 +0ms
  hci write acl data pkt - writing: 0244000b000700060002030001100001 +1ms
  hci onSocketData: 0244000b00070006000204000d100705 +1ms
  hci 	event type = 2 +0ms
  hci onSocketData: 0244201b00410006000cd7169839d1cfbd25e0e6b687c957121298ed2d2db646 +267ms
  hci 	event type = 2 +0ms
  hci 		cid = 6 +0ms
  hci onSocketData: 0244101b00c92f47994458dd93a76e47d16c6c21e2125538bc560eb65c47543a +1ms
  hci 	event type = 2 +0ms
  hci onSocketData: 0244100f00a100ff9e80978d8b4d164900aab82d +3ms
  hci 	event type = 2 +0ms
  hci onSocketData: 0244004500410006000c2a067c0919e4486ddfa38394731afbe4b59cd310892ac868d9b664ba3104e7eb66923c24e2eacbaf6f455aa12208a33fe0aa27678eea7c6959fd90b41e5539c7 +26ms
  hci 	event type = 2 +0ms
  hci onSocketData: 02440015001100060003b343b335f3ca1b28835f110179a1b665 +0ms
  hci 	event type = 2 +1ms
  hci onSocketData: 02442015001100060004b888c57b899b65d322682433df34bbed +172ms
  hci 	event type = 2 +0ms
  hci 		cid = 6 +1ms
  hci 		handle = 68 +0ms
  hci 		data = 04b888c57b899b65d322682433df34bbed +0ms
  hci set advertise enable - writing: 010a200100 +13ms
  bindings disconnect by server +1ms
  hci disconnect - writing: 01060403440013 +2ms

sandeepmistry, 2016-08-21:
@leducb could you dive into `lib/hci-socket/smp.js` and add some `console.log` statements to determine which of the function arguments is `null`?

leducb, 2016-08-22:
@sandeepmistry  this is the output for handlePairingConfirm : 
bleno - echo
on -> stateChange: poweredOn
on -> advertisingStart: success
on -> accpet: 58:1f:bb:94:33:78
handlePairingrandom data = 1�V���,�h�_�O
buffer creation this._tk = undefined this.r = undefined
 this._iat =  this._ia = x3��X this._rat =  this._ra = q�}

so i think this_tk and this.r are both null but i don't know why. If you have other question i'm very reactive.

thanks again

sandeepmistry, 2016-10-10:
@leducb one more request, can you also provide and HCI dump capture `sudo hcidump -t -x`.

WIStudent, 2016-10-30:
I am getting the exact same error message. Here is my HCI dump:
```
HCI sniffer - Bluetooth packet analyzer ver 5.42
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-10-30 03:50:08.098583 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-10-30 03:50:08.099425 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-10-30 03:50:08.099465 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-10-30 03:50:08.099925 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-10-30 03:50:08.099957 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-10-30 03:50:08.100435 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-10-30 03:50:08.100466 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-10-30 03:50:08.100891 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-10-30 03:50:08.100921 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-10-30 03:50:08.101298 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-10-30 03:50:08.101325 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-10-30 03:50:08.101743 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:A7:1B:9D
2016-10-30 03:50:08.113898 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-10-30 03:50:08.114304 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-10-30 03:50:08.114698 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 100.000ms, max 100.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2016-10-30 03:50:08.115267 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2016-10-30 03:50:08.122624 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2016-10-30 03:50:08.123411 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2016-10-30 03:50:08.123452 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  1B 02 01 06 10 09 50 6F 6B 65 6D 6F 6E 20 47 4F 20 50 6C 75 
  73 06 20 62 04 C5 21 00 00 00 00 00 
2016-10-30 03:50:08.124180 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2016-10-30 03:50:08.124215 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2016-10-30 03:50:08.124678 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2016-10-30 03:50:08.124710 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2016-10-30 03:50:08.125652 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
[... 350 lines cut ...]
    SMP: Pairing Random (0x04)
      random 077013c2d14bc82bde69cc3bcc6a1eba
2016-10-30 03:50:34.578536 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-10-30 03:50:34.578981 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-10-30 03:50:34.580410 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-10-30 03:50:34.580848 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-10-30 03:50:34.656099 > ACL data: handle 64 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0008, end 0x0009
      type-uuid 0x2803
2016-10-30 03:50:34.656354 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-10-30 03:50:34.746927 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-10-30 03:50:34.747044 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

WIStudent, 2016-11-01:
Seeing that there are two Pairing Responses made me suspect that something else besides bleno is trying to answer to the Pairing Request. As in issue #225 suggested I tried using `sudo hciconfig hci0 down` and sudo `HCI_CHANNEL_USER=1 node <file>.js` to prevent the kernel from handling SMP messages. That fixed the initial problem but revealed a new one. Now it fails to respond to the LE Long Term Key Request, as you can see in the following HCI dump. Issue #187 seems to deal with this problem already.

```
> HCI Event: LE Meta Event (0x3e) plen 19                                                                                                                                                   [hci0] 5.098756
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Slave (0x01)
        Peer address type: Random (0x01)
        Peer address: 7C:B4:FA:30:DA:AE (Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0.00 msec (0x0000)
        Supervision timeout: 20000 msec (0x07d0)
        Master clock accuracy: 0x00
> ACL Data RX: Handle 64 flags 0x02 dlen 13                                                                                                                                                 [hci0] 5.346323
      ATT: Find By Type Value Request (0x06) len 8
        Handle range: 0x0001-0xffff
        Attribute type: Primary Service (0x2800)
          UUID: Generic Attribute Profile (0x1801)
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                                                                                                                 [hci0] 5.346879
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey Sign (0x07)
        Responder key distribution: EncKey IdKey Sign (0x07)
< ACL Data TX: Handle 64 flags 0x00 dlen 9                                                                                                                                                  [hci0] 5.355627
      ATT: Find By Type Value Response (0x07) len 4
        Handle range: 0x0006-0x0009
< ACL Data TX: Handle 64 flags 0x00 dlen 11                                                                                                                                                 [hci0] 5.359591
      SMP: Pairing Response (0x02) len 6
        IO capability: NoInputNoOutput (0x03)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, No MITM, Legacy, No Keypresses (0x01)
        Max encryption key size: 16
        Initiator key distribution: <none> (0x00)
        Responder key distribution: EncKey (0x01)
> ACL Data RX: Handle 64 flags 0x02 dlen 13                                                                                                                                                 [hci0] 5.436381
      ATT: Find By Type Value Request (0x06) len 8
        Handle range: 0x000a-0xffff
        Attribute type: Primary Service (0x2800)
          UUID: Generic Attribute Profile (0x1801)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                      [hci0] 5.436613
        Num handles: 1
        Handle: 64
        Count: 2
< ACL Data TX: Handle 64 flags 0x00 dlen 9                                                                                                                                                  [hci0] 5.438303
      ATT: Error Response (0x01) len 4
        Find By Type Value Request (0x06)
        Handle: 0x000a
        Error: Attribute Not Found (0x0a)
> ACL Data RX: Handle 64 flags 0x02 dlen 21                                                                                                                                                 [hci0] 5.481459
      SMP: Pairing Confirm (0x03) len 16
        Confim value: 528b0718d4e83c2ea79fa9e881484ce7
< ACL Data TX: Handle 64 flags 0x00 dlen 21                                                                                                                                                 [hci0] 5.492555
      SMP: Pairing Confirm (0x03) len 16
        Confim value: 06a9aa90088b51b87b493be404b6c5bc
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                                                                                                                 [hci0] 5.526272
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x0001-0x0009
        Attribute type: Characteristic (0x2803)
< ACL Data TX: Handle 64 flags 0x00 dlen 27                                                                                                                                                 [hci0] 5.532378
      ATT: Read By Type Response (0x09) len 22
        Attribute data length: 7
        Attribute data list: 3 entries
        Handle: 0x0002
        Value: 020300002a
        Handle: 0x0004
        Value: 020500012a
[... 164 lines cut ...]
        Handle: 64
        Count: 1
[... cut ...]
In the end I managed to establish a pairing by preventing bleno from handling the SMP messages and letting the kernel handle the whole pairing procedure. In order to do that I commented out the following code in the `smp.js` starting at line 53:
```
  if (SMP_PAIRING_REQUEST === code) {
    this.handlePairingRequest(data);
  } else if (SMP_PAIRING_CONFIRM === code) {
    this.handlePairingConfirm(data);
  } else if (SMP_PAIRING_RANDOM === code) {
    this.handlePairingRandom(data);
  } else if (SMP_PAIRING_FAILED === code) {
    this.handlePairingFailed(data);
}
```
By the way I am using a Raspberry Pi 3 with kernel 4.7.6.

jollobajano, 2018-03-31:
Doing nothing but commenting out that `if`-statement worked for me :/

AndyChoon, 2018-06-25:
@jollobajano Hi sir, excuse me, comment out if statement, will make all characteristics cannot work, if enable bluetoothd, Service cannot work. And I also found Just Work cannot activate the characteristic where I makred Secure.  
For example, this one cannot be read after 'pair' window
var StaticReadOnlyCharacteristic = function() {
  StaticReadOnlyCharacteristic.super_.call(this, {
    uuid: 'fffffffffffffffffffffffffffffff1',
    properties: ['read'],
    value: new Buffer('value'),
    secure: ['read'],

How did you do it?

jollobajano, 2018-06-26:
Cheers @AndyChoon. I don't have access to that machine anymore. I'm sorry I can't help, I just didn't want leave you hanging. I hope WIStudent can help you.

Cheers