noble/bleno #218: Automatic state change to poweredOff after disconnect

kubols, 2016-08-24:
Hi Sandeepmistry! I love your bleno library, it's absolutely great. 
I have a small problem and I can't figure it out how to solve it. I have the Raspberry Pi set up as a BLE slave and I'm subscribing to notifications from it with an Android device. Everything is working well, however after unsubscribing and closing the connection between these two devices and turning some sensors down in `bleno.on('disconnect', function(clientAddress) {}` RPi is changing the bluetooth state to poweredOff automatically after a while but I want it to wait for another connection.

sandeepmistry, 2016-08-28:
Hi @kubols,

Strange, I've never seen this. Could you please provide the following:
 * debug logs: `sudo DEBUG=* node <file>.js`
 * HCI dump capture: `sudo hcidump -t -x`
 * Which Bluetooth adapter you are using

sandeepmistry, 2016-09-11:
@kubols ping ...

sandeepmistry, 2016-09-17:
Closing this due to lack of activity.

closed 2016-09-17 by sandeepmistry

dneykov, 2016-11-17:
I have similar problem also on Raspberry Pi. When I make connection with one device everything is working but when I try to connect another device after that it connect to my raspberry pi but right after connection bleno library says that bluetooth is poweredOff

dneykov, 2016-11-17:
Sorry about my previous post. There is no actual problem. My problem was that I had my app running once as a service and one more time manually which caused bluetooth problems.