noble/bleno #220: Retrieving a Descriptor Value - BluetoothGattDescriptor getValue() returns null object from test.js fff1 descriptor value

GrahamHorne, 2016-08-25:
Hi Sandeep, 

Excellent code / BLE implementation.  One issue that I have found is calling the Descriptor Value.  Using your test.js  programme the fff1 characteristic has a descriptor uudi set as "2901" and a value as "user description".  In my Android programme that communicates with the BLE device (running the test.js) I can access the UUID using the standard getUuid() method (from the BluetoothGattDescriptor class), however the value set in test.js is not accessible using the getValue() method (https://developer.android.com/reference/android/bluetooth/BluetoothGattDescriptor.html#getValue()).

I'm just wondering if this is an issue you have come across?  

Many thanks

Graham

sandeepmistry, 2016-08-28:
@GrahamHorne I haven't use the BLE API's on Android too much. Maybe @don has some tips.

I would suggest you try it with the "nRF Master Control" Android application, just to make sure it can read it.

sandeepmistry, 2016-09-26:
I'm closing this due to lack of activity.

closed 2016-09-26 by sandeepmistry