noble/bleno #221: Disconnect after read/write secure characteristic

TheClownDE, 2016-08-26:
Hi,
I'm facing the same issue as in [Issue 71](https://github.com/sandeepmistry/bleno/issues/71) and tried all the mentioned solutions.
Currently I am working withe BlueZ 5.41 (via Homepage), node v4.5.0 and noble/bleno v1.6.
I am able to read and write to my on application (peripheral on Linux Mint 17.2) with my smartphone (Android, nRF Connect app) but as soon as I add the "secure" filed in the characteristic the devices disconnect on write/read request.
The also disconnect on bonding.
Any Solutions?

Thanks for any help,
TheClown

sandeepmistry, 2016-08-28:
Hi @TheClownDE,
Could you please provide the following?

 * which kernel version you are using? (`uname -a`)
 * debug logs: `sudo DEBUG=* node <file>.js`
 * HCI dump capture: `sudo hcidump -t -x`

TheClownDE, 2016-08-29:
Hi @sandeepmistry,
Thank you for your help and your work, great job!

Kernel:
`Linux 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`

Debug Logs:
`  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('42616420454e495341204c6f636b', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +0ms
  bleno platform linux +15ms
[... 660 lines cut ...]
    Reason: Connection Terminated by Local Host
2016-08-29 09:33:07.730596 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 1280.000ms, max 1280.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2016-08-29 09:33:07.745444 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2016-08-29 09:33:07.745539 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2016-08-29 09:33:07.750472 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2016-08-29 09:33:07.750565 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2016-08-29 09:33:07.754405 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-08-29 09:33:11.658418 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-08-29 09:33:11.660390 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00`

sandeepmistry, 2016-09-05:
Hi @TheClownDE,

Looks like the kernel is sending a disconnect:

```
2016-08-29 09:33:07.654172 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 68 reason 0x13
    Reason: Remote User Terminated Connection
```

Is there any way for you to try a newer kernel? My Raspberry Pi is running ` 4.1.19-v7+` so `3.16.0-38` seems very old.

TheClownDE, 2016-09-06:
Hi Sandeepmistry,

Thanks for your help.
I updated my Linux Mint 17.2 to Kernel 4.4.0-35-generic but I still have the same issue.

Just for clarification:
I added
`secure: [ 'read, 'write' ]`
to my characteristic and tried to send a read request, which fails and leads to a disconnect.

Kind regards,
TheClown