noble/bleno #222: Primary Service advertised but not discoverable

relevante, 2016-08-26:
Hi,

There seems to be an issue with bleno on the nextthing.co CHIP. I just confirmed that the same issue affecting my code is also affecting the pizza example.

Specifically, the primary service is advertised, but when I connect to the peripheral (using CoreBluetooth on iOS) and discover services, it finds 1803, 1804, 1802, 1805, 1806, 180E, and 1811 and fails to find the 13333333333333333333333333333337 uuid. The example works perfectly on my RPi 3.

CHIP came with bluez 5.23 installed which is what I initially tried, but the behavior is the same with bluez 4.101.

Please let me know how I can help debug. I'm not at all familiar with the underlying libraries, but I'm happy to do whatever I can. I'm also happy to send you a CHIP if you don't have one.

Thanks!

sandeepmistry, 2016-08-28:
Hi @relevante,

I think the problem is that `bluetoothd` is still running the 5.x version. Have you tried stopping it as per https://github.com/sandeepmistry/bleno#linux ? See #24 for more details on why it must be stopped.

I'm closing this for now, because I think this is the problem. We can re-open if it's not the case.

closed 2016-08-28 by sandeepmistry

relevante, 2016-08-28:
Hi @sandeepmistry,

Thanks for pointing this out. I guess maybe the old version was running and didn't get shutdown because I had installed the new one over the top or something. I was aware of the need to shut it down and thought I had, but it's working now after a reboot and ensuring it's shut down. Thanks for the help (and for the awesome work in general!).

sandeepmistry, 2016-09-05:
@relevante cool! Great to hear that you got things working.