noble/bleno #223: Characteristics not listed when connecting to Service from a Central

goability, 2016-08-28:
Using Intel Edison and sample example, I am unable to see a list of characteristics.  My goal is to see the characteristic and set its value using any simple free mobile app (nRF Connect, BLE Scanner, any CySmart all produce same results).  

Thanks so much in advance for any tips, I'm sure it's something simple I'm doing wrong..

Example code I took from Bleno example, and even tried to add descriptor, however no success:

//In characteristic.js, I define the char..

var FirstCharacteristic = function() {
  FirstCharacteristic.super_.call(this, {
    uuid: 'fff1',
    properties: ['read', 'write', 'notify'],
    value: 'Light1',
    descriptors: [new bleno.Descriptor({uuid:'2901',value:'Light1'})]
  });

//In main.js I set the service and reference the characteristics I created.
bleno.setServices([
      new BlenoPrimaryService({
        uuid: 'fff0',
        characteristics: [
          new FirstCharacteristic(),
          new SecondCharacteristic()
        ]
      })

atilag, 2016-08-29:
Got the same problem :(

atilag, 2016-08-30:
So I was just trying to run both **test.js** present in [bleno](https://github.com/sandeepmistry/bleno/blob/master/test.js) and [noble](https://github.com/sandeepmistry/noble/blob/master/test.js) modules. I have a RPI2 with Raspbian (Linux 4.1.13) that has a CSL USB Nano Bluetooth Dongle (CSR8510 A10) attached, this is where I executed bleno test.js (bleno-test.log) and on my laptop I run noble test.js (noble-test.log). The problem is when the "servicesDiscover" handler is executed, because services parameter is empty, so few lines below the noble test.js program crashes:

```
peripheral.on('servicesDiscover', function(services) {
    ... 
    // crashes here because 'services' is empty, so 'services [serviceIndex]' is out of bounds...
    services[serviceIndex].on('includedServicesDiscover', function(includedServiceUuids) {
   ...
}
```
If I run nRF Connect Android App and try to get a list of the Services/Characteristics, it doesn't retrieve anything either.
I have attached the logs of the test.js runs just in case, if you need a hcidump output too, just tell me and I will provide it.

[bleno-noble-test-logs.zip](https://github.com/sandeepmistry/bleno/files/446066/bleno-noble-test-logs.zip)

atilag, 2016-08-31:
Ok, I could make it work with the RPI3 in-built chipset, so the problem above must be related to unsupported HCI commands for that chipset (CSR8510 A10).

goability, 2016-08-31:
I have it working now on Edison as well by redoing sample code and reflashing the image.  Not clear on root cause, but there is no issue anymore.

sandeepmistry, 2016-09-05:
@goability thanks for the update, I'm going to close this for now them.

@atilag please open a new issue with the HCI dump. Maybe we need a max MTU workaround like the Realtek? https://github.com/sandeepmistry/bleno/blob/95e33c202ab2b9506f1d6932413adcf709d5d7fa/lib/hci-socket/bindings.js#L124-L127

closed 2016-09-05 by sandeepmistry