noble/bleno #225: Solved my bonding problem

batminseok, 2016-08-30:
In the last week I've been writing a bleno script, and while it always connected and sent/read data fine, it never bonded. After a couple days of exasperation at not finding a solution (I've noticed a lot of people on this issues section have the same problem), I gave up and decided to try writing my very own plugin. It went alright up until I hit a wall (problems with getting my own 128bit uuid's to play nicely with notifications), so I went back and tested bleno again. To my great surprise my bleno service was bonded! And even better, it was sending notifications. 

The bluez plugin tutorial and script came from here: 
http://blog.csdn.net/u013606170/article/details/47093773

I'm using bluez 5.28, the latest bleno, and the devices it was tested on were android 5.0.2 using nRF Connect and android 6.0.1 using BLE Scanner.

After trying to replicate this behaviour for a while I figured out what I had done: 
1.  turned hci0 to up
2.  ran my bleno script
3.  ran the bleAdvertise.sh script described in the link above
4.  stopped bleno, and the bluetooth service
5. ran the wiimote script from the above link 
6. ran the bleAdvertise script again. 

There's probably a few steps I could cut out/change but this must be run if I want to initially bond a device. The plugin script is run and bonds with the device, somehow displaying bleno characteristics. Secure characteristics still don't work, and there are still issues with the process either displaying the wiimote characteristics or the default bluez service (time/alert).

While this is useful, is there any easier and more reliable way to get bleno bonding?

sandeepmistry, 2016-09-05:
Hi @batminseok,

Thanks for sharing. I think the best way would be to stop relying on the kernel's SMP stack for pairing. This can be be done via `sudo hciconfig hci0 down` and `sudo HCI_CHANNEL_USER=1 node <file>.js`. Then the built-in SMP stack in bleno needs to filled in.

I'm closing this for now, since there are multiple issues open regarding pairing. If you have time to try the above out and send a PR that would be awesome.

closed 2016-09-05 by sandeepmistry

batminseok, 2016-09-15:
Ran my bleno script with the HCI_CHANNEL_USER=1 command and still exhibits the same behaviour.

Been a while but I ran btmon on bonding attempts between the wiimote gatt plugin and the bleno script (still not bonding) that I wrote.

The wiimote example, when connecting from the same tablet, bonds straight away:

`@ Device Connected: 34:14:5F:85:7B:D6 (1) flags 0x0000 //so pretty much immediately 
> HCI Event: Command Status (0x0f) plen 4                      
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> ACL Data RX: Handle 64 flags 0x02 dlen 11                
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0001-0xffff
        Attribute group type: Primary Service (0x2800)
> HCI Event: LE Meta Event (0x3e) plen 12                      [hci0] 58.538351
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 64
        Features: 0x0f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
< ACL Data TX: Handle 64 flags 0x00 dlen 16                    [hci0] 58.538504
      LE L2CAP: Connection Parameter Update Request (0x12) ident 1 len 8
        Min interval: 40
        Max interval: 56
        Slave latency: 0
        Timeout multiplier: 2000
< ACL Data TX: Handle 64 flags 0x00 dlen 6                     [hci0] 58.538805
      SMP: Security Request (0x0b) len 1 //smp request transmitted
        Authentication requirement: Bonding, No MITM, SC, No Keypresses (0x09)
> ACL Data RX: Handle 64 flags 0x02 dlen 10                    [hci0] 58.830546
      LE L2CAP: Connection Parameter Update Response (0x13) ident 1 len 2
        Result: Connection Parameters accepted (0x0000) //LE connection update accepted
> HCI Event: Number of Completed Packets (0x13) plen 5         [hci0] 58.830746
        Num handles: 1
        Handle: 64
        Count: 2
> ACL Data RX: Handle 64 flags 0x02 dlen 11                    [hci0] 58.830987
      SMP: Pairing Request (0x01) len 6 //receives request from the tablet
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey Sign LinkKey (0x0f)
        Responder key distribution: EncKey IdKey Sign LinkKey (0x0f)
< ACL Data TX: Handle 64 flags 0x00 dlen 11                    [hci0] 58.831045
      SMP: Pairing Response (0x02) len 6 //sends smp response
        IO capability: NoInputNoOutput (0x03)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey Sign LinkKey (0x0f)
        Responder key distribution: EncKey Sign LinkKey (0x0d)
> ACL Data RX: Handle 64 flags 0x02 dlen 27                    [hci0] 58.928290
> ACL Data RX: Handle 64 flags 0x01 dlen 27                    [hci0] 58.928934
> ACL Data RX: Handle 64 flags 0x01 dlen 15                    [hci0] 58.929409
      SMP: Pairing Public Key (0x0c) len 64 //keys exchange
        X: 7ef9d89292bfc36d3235741ce11da5542b5f298ba3e21ab0a488d9ebccd48557
        Y: 2369a09c6ab31258964afe768eb8129bab078915a902a8512a5e1f79c35565a2
< ACL Data TX: Handle 64 flags 0x00 dlen 69                    [hci0] 58.943414
      SMP: Pairing Public Key (0x0c) len 64 //more keys exchanging
        X: 9678be8131d5e7526b9a44c1d2426e350260083d043f30ac5fcfc7334ecfc1f2
        Y: 5ded2a45024bc3a256fea9d209c8c174a237eca44be08518d7973c746ed15b9a
< ACL Data TX: Handle 64 flags 0x00 dlen 21                    [hci0] 58.943441
      SMP: Pairing Confirm (0x03) len 16 //pairing confirmed!
        Confim value: e7dcffcd463c67e8103855fcb44806ef 
> HCI Event: Number of Completed Packets (0x13) plen 5         [hci0] 58.978613
        Num handles: 1
[... 171 lines cut ...]
      LE Long Term Key Request (0x05)
        Handle: 64
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
< HCI Command: LE Long Term Key Reque.. (0x08|0x001b) plen 2  [hci0] 312.946432
        Handle: 64
> HCI Event: Command Complete (0x0e) plen 6                   [hci0] 312.946854
      LE Long Term Key Request Neg Reply (0x08|0x001b) ncmd 1
        Status: Success (0x00)
        Handle: 64
< ACL Data TX: Handle 64 flags 0x00 dlen 6                    [hci0] 312.947606
      SMP: Pairing Failed (0x05) len 1
        Reason: Unspecified reason (0x08)
> HCI Event: Number of Completed Packets (0x13) plen 5        [hci0] 313.193662
        Num handles: 1
        Handle: 64
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5        [hci0] 313.568620
        Num handles: 1
        Handle: 64
        Count: 1`

While I havent really looked at any of the mechanisms behind bonding and encryption this quick look shows me that long term keys arent exchanged between the bleno script and the tablet on connection, stopping the bonding. The SMP request is sent from the pi on the wiimote, and from the tablet on the bleno. 

I havent logged  the HCI_CHANNEL_USER=1 method yet but I can post that as well.

batminseok, 2016-09-15:
Connection is a bit different with the HCI_CHANNEL_USER=1 command: 

`> HCI Event: LE Meta Event (0x3e) plen 19                     [hci0] 138.624394
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Slave (0x01)
        Peer address type: Random (0x01)
        Peer address: 58:18:3B:86:13:A4 (Resolvable)
        Connection interval: 48.75 msec (0x0027)
        Connection latency: 0.00 msec (0x0000)
        Supervision timeout: 20000 msec (0x07d0)
        Master clock accuracy: 0x01
> ACL Data RX: Handle 64 flags 0x02 dlen 11                   [hci0] 138.892090
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0001-0xffff
        Attribute group type: Primary Service (0x2800)`

and then skips straight to reading services, no @device connected line.

Bonding request from the tablet is similar enough though: 

`> ACL Data RX: Handle 64 flags 0x02 dlen 11                   [hci0] 236.265490
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey Sign LinkKey (0x0f)
        Responder key distribution: EncKey IdKey Sign LinkKey (0x0f)
< ACL Data TX: Handle 64 flags 0x00 dlen 11                   [hci0] 236.267115
      SMP: Pairing Response (0x02) len 6
        IO capability: NoInputNoOutput (0x03)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, No MITM, Legacy, No Keypresses (0x01)
        Max encryption key size: 16
        Initiator key distribution: <none> (0x00)
        Responder key distribution: EncKey (0x01)
> ACL Data RX: Handle 64 flags 0x02 dlen 21                   [hci0] 236.363241
      SMP: Pairing Confirm (0x03) len 16
        Confim value: 4d093077d85806c96873df2fec8d5547
< ACL Data TX: Handle 64 flags 0x00 dlen 21                   [hci0] 236.372681
      SMP: Pairing Confirm (0x03) len 16
        Confim value: 340345024340bc9841baea2277c780b9
> ACL Data RX: Handle 64 flags 0x02 dlen 21                   [hci0] 236.460781
      SMP: Pairing Random (0x04) len 16
        Random value: fc053491780eb0c260afe6daa5ed62d5
> HCI Event: Number of Completed Packets (0x13) plen 5        [hci0] 236.460931
        Num handles: 1
        Handle: 64
        Count: 2
< ACL Data TX: Handle 64 flags 0x00 dlen 21                   [hci0] 236.466398
      SMP: Pairing Random (0x04) len 16
        Random value: 6a2cadcfaa56bacf038cc544d342f369
> HCI Event: LE Meta Event (0x3e) plen 13                     [hci0] 236.558646
      LE Long Term Key Request (0x05)
        Handle: 64
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
> HCI Event: Number of Completed Packets (0x13) plen 5        [hci0] 236.695506
        Num handles: 1
        Handle: 64
        Count: 1`

sandeepmistry, 2016-09-17:
Hi @batminseok,

Thanks for the update! You might also find this page of use: https://community.nxp.com/thread/332191

I used it as a reference when developing the noble and bleno SMP stacks.