noble/bleno #234: Could not initialize libusb. Check that your system has a USB controller.

sdurandeu, 2016-10-03:
I'm using the library in Windows 7 SP1 32 bits. I've enabled libusb debug (using `set LIBUSB_DEBUG=4`) and it seems the underlying error is:

`[ 0.002930] [00001208] libusb: error [windows_hotplug_threaded] can't register hotplug message window class [1410] Class already exists.`

The BLE device is a CSR8510 A10. All suggestions are welcomed :)

Thanks!
Sebastian

sandeepmistry, 2016-10-10:
Hi @sdurandeu,

Did you use Zadig tool to unload the default Windows driver for the adapter? There's a video by @don  for noble on the related steps: https://www.youtube.com/watch?v=mL9B8wuEdms

sdurandeu, 2016-10-11:
Thanks @sandeepmistry. I was actually using [node-bleacon](https://github.com/sandeepmistry/node-bleacon). However as I only need advertising, I ended up using just bleno. This way the error disappeared, so it was probably related to bleacon. 

I will close this issue.

closed 2016-10-11 by sdurandeu