noble/bleno #235: Change connection interval ?

natcl, 2016-10-09:
Hello,
I'm trying to implement the Apple [BLE MIDI Service](https://developer.apple.com/bluetooth/Apple-Bluetooth-Low-Energy-MIDI-Specification.pdf).

In the documentation they state this:
> The accessory shall request a connection interval of 15 ms or less. Apple recommends starting with a request for a connection interval of 11.25 ms and going to 15 ms if the connection request is rejected by the Apple product.

Is there a way to set this in bleno ?

Thanks

sandeepmistry, 2016-10-10:
Hi @natcl,

This is not supported at the moment, I'm marking it as an enhancement.

We could have an API similar to CoreBluetooth's [setDesiredConnectionLatency](https://developer.apple.com/reference/corebluetooth/cbperipheralmanager/1393277-setdesiredconnectionlatency) with values low, medium, and high.

natcl, 2016-10-10:
Thanks, and what would be the default connection interval now ? Is it the same as the 100ms advertising interval or once a connection is established it's faster ?

sandeepmistry, 2016-10-10:
I believe it's determined by the central device, you can use `sudo hcidump -t -x` to see what is specified. Please let me know what it reports on your system.

natcl, 2016-10-10:
In my case the central device is my iPhone I believe, the peripheral (using
bleno) runs on MacOS.

sandeepmistry, 2016-10-10:
You can use the Packet Logger app that is part of Xcode Hardware IO tools to see what's going on then.

natcl, 2016-10-10:
Thanks will try that !
So far speed seems good so I assume nothing more is needed in bleno.

sandeepmistry, 2016-10-10:
@natcl excellent, let's leave this request open for now to see if other people request a similar feature.

iankchristie, 2016-12-08:
Hello!

I'm working on transmitting small files from a Raspberry Pi to an Android. Being able to increase the connection interval would be extremely helpful to speed up transfer.

rzr, 2020-11-04:
Relate-to:

https://github.com/abandonware/bleno/issues/17