noble/bleno #240: Discovery of services and characteristics on iPhone does not work

Cheesebaron, 2016-10-31:
I have made an interface with 6 services, one of them being the Device Information service.

I have 23 characteristics split over the 6 services, the service with the most characteristics has 8 of them.

Android devices, which typically report an MTU of 23 discover all services and characteristics fine. If I use an iPhone 6, which appears with MTU 185, no services or characteristics show up.

I've messed around a bit inside of [gatt.js](https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js), where I've discovered when setting `numServices` and `numCharacteristics` to 1

Here
https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L559

and here
https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L615

then the services and characteristics show up fine. This is similar to fooling the bleno to think the device connects with a different MTU.

Why is it that devices with high MTU won't discover any of my services and characteristics?

sandeepmistry, 2016-10-31:
@Cheesebaron please provide a sample bleno app that can be used to reproduce the issue.

Cheesebaron, 2016-11-10:
After more debugging it seems to be related to the Intel 7260 HMWWB mini-PCIe module we are using. Running the code on a RPi or macOS works fine.

```
var bleno = require('bleno');
var PrimaryService = bleno.PrimaryService;
var Characteristic = bleno.Characteristic;
var Descriptor = bleno.Descriptor;

function start() {
    var namespaceStart = 'c3daa788';
	var namespaceEnd = '545aad2a3c51';

	var deviceIdentifier = '776874737172';
    var uuid = namespaceStart + deviceIdentifier + namespaceEnd;

	console.log('namespace start: \'' + namespaceStart + '\'');
	console.log('namespace end: \'' + namespaceEnd + '\'');
	console.log('deviceId: \'' + deviceIdentifier + '\'');
	console.log('uuid: \'' + uuid + '\'');

    var name = "Bleno derp";
    process.env.BLENO_DEVICE_NAME = name;
    process.env.BLENO_ADVERTISING_INTERVAL = 100;

    console.log('Waiting for bluetooth...');

    var batteryService = new PrimaryService({
        uuid: '180f',
        characteristics: [
            new Characteristic({
                uuid: '2a19',
                properties: ['read'],
                descriptors: [
                    new bleno.Descriptor({
                        uuid: '2901',
                        value: 'Battery level'
                    })
                ],
                value: Buffer.from('100')
            })
        ]
    });

    var deviceInfoService = new PrimaryService({
        uuid: '180a',
        characteristics: [
            new Characteristic({
                uuid: '2a25',
                properties: ['read'],
                descriptors: [
                    new bleno.Descriptor({
                        uuid: '2901',
                        value: 'Serial number'
                    })
                ],
                value: Buffer.from('12345678')
            }),
            new Characteristic({
                uuid: '2a29',
                properties: ['read'],
                descriptors: [
                    new bleno.Descriptor({
                        uuid: '2901',
                        value: 'Manufacturer name'
                    })
                ],
                value: Buffer.from('Derp')
            }),
            new Characteristic({
                uuid: '2a24',
[... 335 lines cut ...]
            });
        } else {
            bleno.stopAdvertising();
        }
    });

    bleno.on('advertisingStart', function(err) {
        if (!err) {
            console.log('Started advertising with uuid', uuid);

            console.log('Setting services...');
            bleno.setServices([
                batteryService,
                deviceInfoService,
                moreInfoService,
                simConfigurationService,
                apConfigurationService,
                wifiConfigurationService
            ]);
        }
    });
}

start();
```

sandeepmistry, 2016-11-14:
@Cheesebaron can you please provide an HCI dump capture (`sudo hcidump -t -x`)?

Maybe that specific adapter doesn't support higher MTU's?

Cheesebaron, 2016-11-14:
Two dumps here. The first is the one where no mods are done to gatt.js. The second is when forcing MTU to 23.

```
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-11-14 10:07:37.435935 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 3585, role slave
      bdaddr 6E:D4:D4:F0:0B:FC (Random)
2016-11-14 10:07:37.549930 > ACL data: handle 3585 flags 0x02 dlen 11
    L2CAP(d): cid 0x003a len 7 [psm 0]
      09 05 02 00 00 00 00 
2016-11-14 10:07:37.550430 > ACL data: handle 3585 flags 0x02 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 185
2016-11-14 10:07:37.553112 < ACL data: handle 3585 flags 0x00 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 185
2016-11-14 10:07:37.610040 > ACL data: handle 3585 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2016-11-14 10:07:37.610922 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:07:37.612038 < ACL data: handle 3585 flags 0x00 dlen 30
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0005
      value 0x00 0x18
      attr handle 0x0006, end group handle 0x0009
      value 0x01 0x18
      attr handle 0x000a, end group handle 0x001c
      value 0x0a 0x18
      attr handle 0x001d, end group handle 0x0021
      value 0x0f 0x18
2016-11-14 10:07:37.670035 > ACL data: handle 3585 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0022, end 0xffff
      type-uuid 0x2800
2016-11-14 10:07:37.670916 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:07:37.673891 < ACL data: handle 3585 flags 0x00 dlen 106
    ATT: Read By Group resp (0x11)
      attr handle 0x0022, end group handle 0x0028
      value 0x0a 0x18 0x2a 0xcd 0x5c 0x54 0x72 0x71 0x73 0x74 0x68 0x77 0x99 0xc7 0xdc 0xc3
      attr handle 0x0029, end group handle 0x0038
      value 0x08 0x94 0x2a 0xcd 0x5c 0x54 0x72 0x71 0x73 0x74 0x68 0x77 0x99 0xc7 0xdc 0xc3
      attr handle 0x0039, end group handle 0x0042
      value 0x08 0x4f 0x2a 0xcd 0x5c 0x54 0x72 0x71 0x73 0x74 0x68 0x77 0x99 0xc7 0xdc 0xc3
      attr handle 0x0043, end group handle 0x004c
      value 0x5e 0x61 0x2a 0xcd 0x5c 0x54 0x72 0x71 0x73 0x74 0x68 0x77 0x99 0xc7 0xdc 0xc3
      attr handle 0x004d, end group handle 0x0062
      value 0xe6 0x40 0x2a 0xcd 0x5c 0x54 0x72 0x71 0x73 0x74 0x68 0x77 0x99 0xc7 0xdc 0xc3
2016-11-14 10:07:37.730912 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:07:52.010665 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 3585 reason 0x13
    Reason: Remote User Terminated Connection
2016-11-14 10:07:52.104242 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2016-11-14 10:07:52.105640 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 2
    status 0x00
```

---------------------

```
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-11-14 10:04:26.025057 > HCI Event: LE Meta Event (0x3e) plen 19
[... 887 lines cut ...]
      handle 0x005c
2016-11-14 10:04:32.261687 < ACL data: handle 3585 flags 0x00 dlen 17
    ATT: Read resp (0x0b)
      57 69 46 69 20 4E 65 74 6D 61 73 6B 
2016-11-14 10:04:32.261573 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:04:32.320342 > ACL data: handle 3585 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x005f
2016-11-14 10:04:32.321273 < ACL data: handle 3585 flags 0x00 dlen 17
    ATT: Read resp (0x0b)
      57 69 46 69 20 47 61 74 65 77 61 79 
2016-11-14 10:04:32.321466 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:04:32.380329 > ACL data: handle 3585 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x0062
2016-11-14 10:04:32.381168 < ACL data: handle 3585 flags 0x00 dlen 13
    ATT: Read resp (0x0b)
      57 69 46 69 20 44 4E 53 
2016-11-14 10:04:32.381339 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-14 10:04:32.440457 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
```

sandeepmistry, 2016-11-16:
I suggest we add a workaround to lower the MTU like I did for RealTek chips here: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/bindings.js#L123-L128

Please submit a pull request for the change.

Cheesebaron, 2016-11-17:
I get this when I interrogate the arguments in that method:

    hciVer: 6 hciRev: 3584 lmpVer: 6 manufacturer:  lmpSubVer: 3584

Manufacturer doesn't seem to be set, what do you recommend using to identify it?

sandeepmistry, 2016-11-19:
Can you please provide a full HCI dump from when you start your bleno app? This will show us what the adapter is sending up.

Cheesebaron, 2016-11-20:
Here you go:

```
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2016-11-20 16:33:19.692799 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-11-20 16:33:19.693982 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-11-20 16:33:19.694019 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-11-20 16:33:19.694982 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-11-20 16:33:19.695029 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-11-20 16:33:19.695984 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0xe00
    LMP Version: 4.0 (0x6) LMP Subversion: 0xe00
    Manufacturer: Intel Corp. (2)
2016-11-20 16:33:19.696032 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-11-20 16:33:19.696976 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-11-20 16:33:19.697008 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-11-20 16:33:19.697982 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-11-20 16:33:19.698032 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-11-20 16:33:19.698987 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 7C:5C:F8:D4:98:BE
2016-11-20 16:33:19.728974 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-11-20 16:33:19.729937 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-11-20 16:33:19.731140 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 100.000ms, max 100.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2016-11-20 16:33:19.731933 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2016-11-20 16:33:19.736781 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  0C 0B 08 42 6C 65 6E 6F 20 64 65 72 70 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2016-11-20 16:33:19.738464 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2016-11-20 16:33:19.738591 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 51 3C 2A AD 5A 54 72 71 73 74 68 77 88 A7 
  DA C3 00 00 00 00 00 00 00 00 00 00 
2016-11-20 16:33:19.739067 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2016-11-20 16:33:19.739106 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2016-11-20 16:33:19.740308 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 2
    status 0x00
2016-11-20 16:33:19.740449 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  0C 0B 08 42 6C 65 6E 6F 20 64 65 72 70 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2016-11-20 16:33:19.740983 > HCI Event: Command Complete (0x0e) plen 4
[... 715 lines cut ...]
2016-11-20 16:33:57.576873 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-20 16:33:57.584267 > ACL data: handle 3585 flags 0x02 dlen 9
    ATT: Find Information req (0x04)
      start 0x004e, end 0x004e
2016-11-20 16:33:57.585110 < ACL data: handle 3585 flags 0x00 dlen 10
    ATT: Find Information resp (0x05)
      format: uuid-16
        handle 0x004e, uuid 0x2901 (GATT(desc) User Description)
2016-11-20 16:33:57.599874 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-20 16:33:57.606556 > ACL data: handle 3585 flags 0x02 dlen 9
    ATT: Find Information req (0x04)
      start 0x0051, end 0x0051
2016-11-20 16:33:57.615698 < ACL data: handle 3585 flags 0x00 dlen 10
    ATT: Find Information resp (0x05)
      format: uuid-16
        handle 0x0051, uuid 0x2901 (GATT(desc) User Description)
2016-11-20 16:33:57.629880 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
2016-11-20 16:33:57.831157 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 3585
      interval 65.00ms, latency 0.00ms, superv. timeout 20000.00ms
```

sandeepmistry, 2016-11-20:
Thanks

```
2016-11-20 16:33:19.695984 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0xe00
    LMP Version: 4.0 (0x6) LMP Subversion: 0xe00
    Manufacturer: Intel Corp. (2)
```

Based on the manufacturer value is `2`. Did you get a chance to look into the type of the `manufacturer` variable?

Cheesebaron, 2016-11-20:
Yeah it is detected correctly. I just messed up the console.log. So I get 2 in onReadLocalVersion too.

I'll make a Pull Request. Are you sure it is OK to just assume that all Intel Bluetooth cards have this problem? Just like you've done with the Realtek ones?

sandeepmistry, 2016-11-20:
> Are you sure it is OK to just assume that all Intel Bluetooth cards have this problem? Just like you've done with the Realtek ones?

I think it's ok for now - it will save other people headaches, unless we find other Intel adapters to test with.

Do you have any other suggestions? Maybe you look to see if there are any HCI commands to get the specific model of the adapter ...

Cheesebaron, 2016-11-20:
I can't seem to find any HCI command that can provide more info apart from HCI revision, which isn't very useful.

sandeepmistry, 2016-12-04:
Closed via #246.

closed 2016-12-04 by sandeepmistry