noble/bleno #244: How to set Advertising non connectable?

TomThats, 2016-11-03:
Hi sandeepmistry,

I'm still a big fan of your  framework, well done!

I stumbled over a little problem, maybe you can help me with it?

I try to make a i Beacon non connectable. I know there is a function on its own for iBeacons, but I want to learn something by trying to build it from scratch myself :)

This is how I build the advertising data:

`        
        // advertisementData
        var advertisementData = new Buffer(30); // maximum 31 bytes

        // first part
        advertisementData.writeUInt8(0x02, 0); // Number of bytes that follow in first AD structure
        advertisementData.writeUInt8(0x01, 1); // Flags AD type
        advertisementData.writeUInt8(0x1A, 2); // Flags value

        // second and last part
        advertisementData.writeUInt8(0x1a, 3); // Number of bytes that follow in second AD structure
        advertisementData.writeUInt8(0xff, 4); // Manufacturer specific data AD type
        advertisementData.writeUInt8(0x4C, 5); // Company identifier code (0x004C == Apple)
        advertisementData.writeUInt8(0x00, 6); // Company identifier code (0x004C == Apple)
        advertisementData.writeUInt8(0x02, 7); // iBeacon advertisement indicator
        advertisementData.writeUInt8(0x15, 8); // iBeacon advertisement indicator
        uuid.copy(advertisementData, 9); // uuid
        advertisementData.writeUInt16BE(major, 25); // major
        advertisementData.writeUInt16BE(minor, 27); // minor

        advertisementData.writeInt8(measuredPower, 29); // calibrated Tx Power
`
It works fine so far, I can see the beacon and its data in my Android DEV-App (nRF Connect) but I'm able to connect to the device which is something I don't want people to do.

I understood that I set the connectable flag in the PDU Part, which I think is the first part in my code above, or not?

I managed to break my phones Bluetooth-Service so bad I had to do a reboot, after I played with the "Flags AD type" values. But could not find a setting to make it non-connectable.

Any idea? :D

Thank you,

Tom

sandeepmistry, 2016-11-06:
Hi @TomThats,

Thanks for the positive feedback!

For now, you'll have to change the following line: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/hci.js#L269

0x00 is for "Connectable undirected", if you use 0x03 " Non connectable undirected" - I think you will reach you goal.

sandeepmistry, 2016-11-20:
I'm closing this for now, I think you question is answered :)

closed 2016-11-20 by sandeepmistry