noble/bleno #247: Multiple devices connected to bleno ?

SamIot, 2016-11-21:
Hi @sandeepmistry  , thanks again for your very good work !

I would like to know if it's possible to connect 2 devices (any smartphones) on the same Bleno and let them use two different services separately? How would it be possible?

Peripheral : Ubuntu mate for Raspberry Pie 3

sandeepmistry, 2016-12-02:
> I would like to know if it's possible to connect 2 devices (any smartphones) on the same Bleno and let them use two different services separately?

If you use 2 separate processes with 2 bluetooth USB adapters this is possible.

See the read me for how to specific with HCI device to use per process: https://github.com/sandeepmistry/bleno#multiple-adapters

closed 2016-12-02 by sandeepmistry

elovin, 2017-11-02:
@sandeepmistry just to be clear about this, multiple devices connected **to the same service** is no problem correct ? (in the noble readme you say something about ~5 simultaneous connections}