noble/bleno #248: Increase MTU

boaerosuke, 2016-11-23:
How can I increase the value size of a characteristic to more then 23 bytes which is the standard?
Is the size (MTU) capped by the capability of a central device? (in my case iOS)

sandeepmistry, 2016-11-26:
This library will adjust the MTU to whatever the central negotiates.

For long reads, you must use the the `offset` parameter of the callback to supply the correct offset in data.

I'm going to close this for now, if you still face issues please provide a minimal sample bleno app to reproduce the issue along with specific sequence of steps to follow.

closed 2016-11-26 by sandeepmistry

boaerosuke, 2016-11-30:
Thank you very much for your response!
I´am playing with your Bleno example and was wondering how to increase the size of the data that has been read onReadRequest.

You wrote I need to use the offset parameter of the callback to supply the correct offset in data

Do you mean the offset in .onReadRequest?

`EchoCharacteristic.prototype.onReadRequest = function(offset, callback) {
  console.log ('onReadRequest Offset: ' + offset.toString());
  //console.log('EchoCharacteristic - onReadRequest: value = ' + this._value.toString('utf8')); 
  callback(this.RESULT_SUCCESS, this._value);
};`

If this is the offset you mentioned, could you tell me where exactly I can configure or increase it? since at this point the log of the offset provides 0, and changing this to another value inside of that function does not have any effect.

Any ideas?

My Goal is to increase the size of the value that has been read. I hope you could unterstand my issue here :)

Again thx for your stuff!

WayneKeenan, 2016-11-30:
Just a FYI:
The Bluetooth LE Stack on your peripheral AND central devices needs to both support the "BLE 4.2 Data Length Extension" to be able to transfer packets up to around 240 bytes.

You might want to confirm that both your devices do support it.

Just a note, as it's not a perfect world I had to disable the BLE DLE on a peripheral so some older macOS laptops could talk to it. Such are the nature of 'standard extensions' I guess.

boaerosuke, 2016-11-30:
Yes, thx for the info :) I´ll keep that in mind.

Increasing the MTU already increased the amount of data that could be read by my central.
But it is not enough, thats why I try to use the offset stuff to see which effects I can achieve.

sandeepmistry, 2016-12-02:
@boaerosuke please see: https://github.com/sandeepmistry/bleno/blob/master/test.js#L50-L75