noble/bleno #249: "GATT Error Unknown"

poshaughnessy, 2016-11-26:
Hi,
I'm not sure where this issue lie, but I'm running a simple peripheral with bleno which I can connect to from a web app, but as soon as I try to read from or write to my characteristic, I get: 

    (index):1 Uncaught (in promise) DOMException: GATT Error Unknown.

My Node code looks like this:

```
//// SERVICE

var BlenoPrimaryService = bleno.PrimaryService;

var RemoteCharacteristic = require('./remote-characteristic');

function RemoteService() {
  RemoteService.super_.call(this, {
    uuid: '180E',
    characteristics: [
      new RemoteCharacteristic()
    ]
  });
}

util.inherits(RemoteService, BlenoPrimaryService);

module.exports = RemoteService;

//// CHARACTERISTIC

var Descriptor = bleno.Descriptor;
var Characteristic = bleno.Characteristic;

var RemoteCharacteristic = function() {
  RemoteCharacteristic.super_.call(this, {
    uuid: '2A19',
    properties: ['read', 'write'],
    descriptors: [
      new Descriptor({
        uuid: '2901',
        value: 'Remote control'
      })
    ]
  });
};

util.inherits(RemoteCharacteristic, Characteristic);

RemoteCharacteristic.prototype.onReadRequest = function(offset, callback) {
  // return hardcoded value just for testing
  callback(this.RESULT_SUCCESS, new Buffer([98]));
};

RemoteCharacteristic.prototype.onWriteRequest = function(data, offset, withoutResponse, callback) {
  if (offset) {
    callback(this.RESULT_ATTR_NOT_LONG);
  }
  else if (data.length !== 1) {
    callback(this.RESULT_INVALID_ATTRIBUTE_LENGTH);
  }
  else {
    var button = data.getUint8(0);
    console.log('button', button);
    switch (button) {
      case 1:
        console.log('left');
        break;
      case 2:
        console.log('right');
        break;
    }
    callback(this.RESULT_SUCCESS);
  }
};

module.exports = RemoteCharacteristic;
```

My central code - I'm using Web Bluetooth - looks like this:

```
      var serviceUuid = 0x180E;
      var charUuid = 0x2A19;
      var characteristic;

      function connect() {

        navigator.bluetooth.requestDevice({
            filters: [{ services: [serviceUuid] }]
          })
          .then(function(device) {
            return device.gatt.connect();
          })
          .then(function(server) {
            return server.getPrimaryService(serviceUuid);
          })
          .then(function(service) {
            return service.getCharacteristic(charUuid);
          })
          .then(function(char) {
            characteristic = char;
          });
      }

      function write(cmd) {
        characteristic.writeValue(cmd);
      }

      function left() {
        var cmd = new Uint8Array(1);
        cmd[0] = 1;
        write(cmd);
      }

      function right() {
        var cmd = new Uint8Array(1);
        cmd[0] = 2;
        write(cmd);
      }
```

(I've tried just reading a value too but it's the same).

(I also tried the pizza example and got another error with that: https://github.com/strangesast/bleno-web-pizza-example/issues/3) 

Thanks in advance!

sandeepmistry, 2016-11-26:
Hi @poshaughnessy,

Let's take a look at the HCI dump capture: `sudo hcidump -t -x` - this will give us some more info. as to what's going on ...

cc/ @beaufortfrancois

beaufortfrancois, 2016-11-28:
@poshaughnessy Can you share platforms you use for Peripheral and Central?

poshaughnessy, 2016-11-28:
@sandeepmistry Sure, I will try this today.
@beaufortfrancois Sorry I forgot to say! Peripheral (Bleno) was Mac OS v10.11 (Macbook Pro Late 2013) and central was Chrome for Android v54 on Android 6.0.1 (Samsung S7).

poshaughnessy, 2016-11-28:
Any tips how to capture that kind of data on a Mac? 

I found the hidden debugging menu (shift + alt + click Bluetooth tray icon) and have taken a Diagnostics Report, but not sure which files are relevant? Then I tried "Enable Bluetooth logging" but since I turned that on, now I can't even connect to the device: 

    (index):1 Uncaught (in promise) DOMException: Connection failed for unknown reason

I switched logging off again and it's still the same. I'll try restarting!

beaufortfrancois, 2016-11-28:
See https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs

sandeepmistry, 2016-11-29:
@poshaughnessy one more suggestion, have you confirmed a native app like nRF master control panel on Android works with bleno on OS X?

poshaughnessy, 2016-11-29:
Thanks @sandeepmistry. I'd not come across that app before; it looks very useful - better than the others I've tried! I can confirm it's working fine using this app as the central - I can read and write values OK 👍 

My Web Bluetooth central is still getting "Connection failed for unknown reason" though.

Apologies for the lack of info and prior testing with my bug report here, I was rushing due to the hackathon time constraints. Shall we close this issue here then and I'll continue with Web Bluetooth bug filing as per the link from @beaufortfrancois?

sandeepmistry, 2016-11-29:
@poshaughnessy no worries, thanks for trying my suggestion out. Yes, let's close this for now, if there's something wrong when bleno, please re-open again.

closed 2016-11-29 by sandeepmistry