noble/bleno #252: Crash after too many connections

guimier, 2016-11-28:
Got the following error after about 10,000 connections:

```
buffer.js:1028
    throw new TypeError('"value" argument is out of bounds');
    ^

TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1028:11)
    at Buffer.writeUInt16LE (buffer.js:1086:5)
    at Mgmt.write (/…/node_modules/bleno/lib/hci-socket/mgmt.js:80:7)
    at Mgmt.loadLongTermKeys (/…/node_modules/bleno/lib/hci-socket/mgmt.js:66:8)
    at Mgmt.addLongTermKey (/…/node_modules/bleno/lib/hci-socket/mgmt.js:47:8)
    at Smp.handlePairingRandom (/…/node_modules/bleno/lib/hci-socket/smp.js:138:10)
    at Smp.onAclStreamData (/…/node_modules/bleno/lib/hci-socket/smp.js:58:10)
    at emitTwo (events.js:111:20)
    at emit (events.js:191:7)
    at AclStream.push (/…/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/…/node_modules/bleno/lib/hci-socket/bindings.js:198:21)
```

sandeepmistry, 2016-12-02:
@guimier this one is going to be tricky for me to reproduce ...

A few questions for you:

 * What are you using as the central?
 * Can you please provide an HCI dump capture?
 * Do you have `bluetoothd` running - if so which BlueZ version?
 * Do you have any secure characteristics in your bleno app?

guimier, 2016-12-02:
The central is Bluez/gatttool on another Pi/Raspbian. I do not have a dumps and probably won’t have the time now to reproduce. I think bluetoothd was running, version 5.23. I did not configure any secure characteristic. It was running with BLUETOOTH_HCI_SOCKET_FORCE_USB=1 because I did not see characteristics otherwise. 

It was running with DEBUG set, too, and it was logging *really big* buffers (unfortunately I don’t have these any more) just before the crash. Seeing the stack above, I think the buffer became too big and the length variable could not be stored as uint16 in `Mgmt.prototype.write`.

sandeepmistry, 2016-12-05:
@guimier how many Bluetooth adapters do you have on the Pi?

> probably won’t have the time now to reproduce

If it's not easy to reproduce, please close the issue - it doesn't seem like something that is high priority ....

sandeepmistry, 2016-12-18:
Closing for now due to lack of activity.

closed 2016-12-18 by sandeepmistry