noble/bleno #253: Does not exit after SIGINT

michaeldeitcher, 2016-12-07:
When I use bleno, I am unable to to exit my script using CTRL-C. 

It is also not exiting on exceptions that are being thrown. 

Thanks

sandeepmistry, 2016-12-18:
@michaeldeitcher I'm going to need more details on this.

I just tried running `sudo node test.js` (from the root folder of the Github repo) on my Raspberry Pi and then a `sudo kill <pid>`. The process exited cleanly, same thing happens for CTRL-C.

Could you please share some minimal code to reproduce the condition as well as step by step instructions.

michaeldeitcher, 2016-12-19:
Thanks for looking into it @sandeepmistry 

I found the issue running on Raspbian Jessie on a Raspberry Pi 3. sudo node test.js is working fine. So there is probably something else going that is causing my issue.

I've started running my node script as a service and I am able to stop the service with out any issues. So, I'm going to close this issue. Thanks again!

closed 2016-12-19 by michaeldeitcher