noble/bleno #255: Connection refused - PSM not supported

sperok, 2017-01-05:
I am trying to use bleno on an NVidia TX-1 development system and have run into a roadblock trying the examples (pizza) receiving the following "Connection refused - PSM not supported" error:

- bleno is running on an NVidia TX-1 (arm) with Ubuntu 16.04. 
- bleno installed via npm install bleno
- kernel is 3.10.96
- bluetoothd v5.37 (turned off with sudo systemctl down bluetooth, verified with ps)
- node v7.3.0
- npm v3.10.10
- pizza example cloned from github

Connections attempted from Linux laptop and Android phone (Nexus 6 with Android 7.0) both running Google Chrome with web bluetooth enabled using the Pizza example web site. Other ble based examples tried also have same problem.

The webapp can scan and find the advertisement, but the connection attempt always fails with the same error. Pairing is successful and the paired device is persistent in the list of bluetooth devices on the laptop and phone.

```
> ACL Data RX: Handle 11 flags 0x02 dlen 12                                       [hci0] 2017-01-05 11:53:57.584204
      L2CAP: Connection Request (0x02) ident 4 len 4
        PSM: 31 (0x001f)
        Source CID: 73
< ACL Data TX: Handle 11 flags 0x00 dlen 16                                       [hci0] 2017-01-05 11:53:57.584313
      L2CAP: Connection Response (0x03) ident 4 len 8
        Destination CID: 0
        Source CID: 73
        Result: Connection refused - PSM not supported (0x0002)
        Status: No further information available (0x0000)
```
A complete btmon log of the session is below, with a few comments added (search for "#"). The connection attempt starts at time 11:53:57.489606, just before that we can see the advertising info get set. The failure shown above occurs at 11:53:57.584313. 

```

# node pizza started on TX-1 system under test
# sudo hciconfig hci0 down 

< HCI Command: Reset (0x03|0x0003) plen 0                                         [hci0] 2017-01-05 11:52:57.330837
> HCI Event: Command Complete (0x0e) plen 4                                       [hci0] 2017-01-05 11:52:57.830993
      Reset (0x03|0x0003) ncmd 1
        Status: Success (0x00)
@ Class of Device Changed: 0x000000
@ New Settings: 0x02d2
            connectable bondable ssp br/edr le 

# sudo hciconfig hci0 up

< HCI Command: Read Local Supported Features (0x04|0x0003) plen 0                 [hci0] 2017-01-05 11:53:09.213720
> HCI Event: Command Complete (0x0e) plen 12                                      [hci0] 2017-01-05 11:53:09.214512
      Read Local Supported Features (0x04|0x0003) ncmd 1
        Status: Success (0x00)
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Paging parameter negotiation
          Power control
          Transparent synchronous data
          Flow control lag (most significant bit)
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
[... 902 lines cut ...]

# And here is the failure

< ACL Data TX: Handle 11 flags 0x00 dlen 16                                       [hci0] 2017-01-05 11:53:57.584313
      L2CAP: Connection Response (0x03) ident 4 len 8
        Destination CID: 0
        Source CID: 73
        Result: Connection refused - PSM not supported (0x0002)
        Status: No further information available (0x0000)
> HCI Event: Number of Completed Packets (0x13) plen 5                            [hci0] 2017-01-05 11:53:57.832863
        Num handles: 1
        Handle: 11
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                                    [hci0] 2017-01-05 11:54:01.671290
        Status: Success (0x00)
        Handle: 11
        Reason: Remote User Terminated Connection (0x13)
@ Device Disconnected: A4:70:D6:B1:27:C2 (0) reason 3
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                       [hci0] 2017-01-05 11:54:01.770118
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                       [hci0] 2017-01-05 11:54:01.771465
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Command Disallowed (0x0c)

```

sandeepmistry, 2017-01-16:
@sperok bleno does not support PSM connections, only L2CAP CID.

closed 2017-01-16 by sandeepmistry

sperok, 2017-01-16:
How does one get a L2CAP CID connection? We are trying to use the unmodified sample app and every connection attempt just fails with this error? Is there a configuration parameter we are missing somewhere in the stack?

sandeepmistry, 2017-01-30:
@sperok I suggest you try using nRF Master Control panel app on Android.