noble/bleno #257: BLE Disconnection after 5 seconds

vaibhav3002, 2017-01-09:
Hi,
I've used bleno on many boards and it works well. In one of my board, however, there is is connection drop after 5 seconds.

The test sequence is simple, 
node test.js (the one that comes from bleno)
From the smartphone, use BLE scanner app (on android) or NRF Connect (on ios), I'm able to discover the device and then connect to it and even able to read characteristics. But, after 5 seconds, the mobile app shows it disconnected. 

I'm using the test program on the bleno. This is an artik board (armv7l)  where the linux kernel is 4.4.19.

I do not see disconnections when using gatt server program using bluez 5.40. Nor do I see disconnections when I use a library which is inspired from bleno 
https://github.com/paypal/gatt

I'd appreciate any help in this matter. Attaching the node debug logs and hcidump output.

[DebugLogs.txt](https://github.com/sandeepmistry/bleno/files/694130/DebugLogs.txt)
[HCIDump.txt](https://github.com/sandeepmistry/bleno/files/694131/HCIDump.txt)

hsigo, 2017-04-11:
Hi,
We had the similar problem on the arktik platform. The example code in the bleno project gets disconnected as well. Do you know what is causing the problem? Is there a solution or workaround for this? Appreciate the help.
# uname -a
Linux artik 4.4.19-0530GC0F-44F-01Q3 #1 SMP PREEMPT ...

sandeepmistry, 2017-04-17:
So I think the kernel is sending a disconnect out:

```
2017-01-09 12:07:21.699585 < HCI Command: Disconnect (0x01|0x0006) plen 3
2017-01-09 12:07:21.699585 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 128 reason 0x13
    Reason: Remote User Terminated Connection
    handle 128 reason 0x13
    Reason: Remote User Terminated Connection
2017-01-09 12:07:21.703723 > HCI Event: Command Status (0x0f) plen 4
2017-01-09 12:07:21.703723 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2017-01-09 12:07:21.809010 > HCI Event: Disconn Complete (0x05) plen 4
2017-01-09 12:07:21.809010 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 128 reason 0x16
    Reason: Connection Terminated by Local Host
    status 0x00 handle 128 reason 0x16
    Reason: Connection Terminated by Local Ho
```

I don't have an Artik board here to test with, but have you tried to compare the HCI dump captures between `bleno` and `paypal/gatt` to see if there are any noticeable differences? Could be something like the connection interval that's different ...

sandeepmistry, 2018-01-29:
Closing due to lack of activity.

closed 2018-01-29 by sandeepmistry

rzr, 2018-06-27:
Which ARTIK board do you use ? I have a couple of boards to test

vaibhav3002, 2018-06-27:
ARTIK 530. It was working with ARTIK 710 and 520 boards though.