noble/bleno #258: Device Name (PI3 / Raspbian Jessie)

Stan92, 2017-01-11:
Hi,
I've used both 2 solutions from the readme file (https://github.com/sandeepmistry/bleno/blob/master/README.md#set-custom-device-name)

But I always get the hostname instead of the BLENO_DEVICE_NAME variable.

Is it possible to use the custom name instead the hostname with Raspberry?
Thanks

sandeepmistry, 2017-01-16:
@Stan92 how are you setting the `BLENO_DEVICE_NAME` env. var?

Stan92, 2017-01-16:
Hi,
Simply like this :
      process.env['BLENO_DEVICE_NAME'] = 'serial-000000001f4f4961';
I also make tests using the command line as you've mentioned in the documentation, I have the same effect.
I add a log
      console.log('Starting myApp : ' + process.env['BLENO_DEVICE_NAME']);
I get the right env value.

In fact, sometimes it works, sometimes it doesn't (in this case, I get the hostname).
I can't reproduce this behavior.
To be more clear, it's by using my own iOS app or/and LightBlue app that I notice the "random" wrong value.
I reboot several times the PI, still get the "hostname", but, I m not 100% sure, if I reboot my iPhone, it seems I get the expected value... (frankly I don't understand why it will be the cause).

sandeepmistry, 2017-01-30:
@Stan92 please share the entire app, so I can try to reproduce locally.

sandeepmistry, 2017-02-13:
Closing this for now due to lack of activity.

closed 2017-02-13 by sandeepmistry