noble/bleno #259: [Question] Don't install dependencies

6pm, 2017-01-14:
I'm using ubuntu 14. My bluetooth device:
```code
hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: C4:85:08:DD:E0:2C  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING PSCAN 
	RX bytes:79785 acl:0 sco:0 events:13105 errors:0
	TX bytes:375134 acl:0 sco:0 commands:13104 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'ubuntu-0'
	Class: 0x00010c
	Service Classes: Unspecified
	Device Class: Computer, Laptop
	HCI Version: 4.0 (0x6)  Revision: 0x1ebd
	LMP Version: 4.0 (0x6)  Subversion: 0xfc00
	Manufacturer: Intel Corp. (2)
```

After I install dependencies I see this errors, debug log attached:

[npm-debug.txt](https://github.com/sandeepmistry/bleno/files/705651/npm-debug.txt)

I think problem is in 'node-hid' module.

6pm, 2017-01-14:
I solve this problem - need install libusb-1.0-0-dev
Thank you for your package.

sandeepmistry, 2017-01-16:
@6pm thanks for the update, closing this for now.

closed 2017-01-16 by sandeepmistry