noble/bleno #265: Can't get services on iOS and Raspberry Pi

PalmerAL, 2017-01-22:
I'm trying to run Bleno on a Raspberry Pi Zero (kernel version 4.4.21+), with a CSR8510, and connect to it on iOS. LightBlue can detect the device and connect to it, but only the default services (time service, alert notification service, etc.) show up. None of the custom services I have set are detected. If I try to connect to the device on a mac, the same thing happens - LightBlue connects, but doesn't find any services.

Code:

```
var bleno = require('bleno');

var Descriptor = bleno.Descriptor;
var descriptor = new Descriptor({
   uuid: '2901',
   value: 'value'
});
var Characteristic = bleno.Characteristic;
var characteristic = new Characteristic({
     uuid: 'fff1',
     properties: [ 'read' ], //'write', 'writeWithoutResponse'
     value: new Buffer('test'), 
     descriptors: [ descriptor ]
});
var PrimaryService = bleno.PrimaryService;
var primaryService = new PrimaryService({
     uuid: 'fffffffffffffffffffffffffffffff0',
     characteristics: [ characteristic ]
});
var services = [ primaryService ];
bleno.on('advertisingStart', function(error) {
     bleno.setServices( services );
});
bleno.on('stateChange', function(state) {
     console.log('BLE stateChanged to: ' + state);
     if (state === 'poweredOn') {
        bleno.startAdvertising('MyDevice',['fffffffffffffffffffffffffffffff0']);
     } else {
        bleno.stopAdvertising();
     }
});
```

Debug logs: 
[debuglog2.txt](https://github.com/sandeepmistry/bleno/files/722178/debuglog2.txt)

HCI dump: 
[hcidump2.txt](https://github.com/sandeepmistry/bleno/files/722179/hcidump2.txt)

I'm guessing these errors are related:

```
2017-01-22 17:45:28.199270 < ACL data: handle 41 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Invalid handle (1)
      Read req (0x0a) on handle 0x0017
```

But I'm not sure how to fix them. Any ideas? Thank you!

gigafied, 2017-01-27:
@PalmerAL I think I'm also running into this, but seems to affect Debian 14.04 (and probably other x64 platforms as well), not just the pi.

The trick to reproducing is to have either > 1 Characteristic or a Descriptor.

If you remove the Descriptor does it work?

gigafied, 2017-01-27:
@sandeepmistry this also seems to affect the examples that use multiple Characteristics or Descriptors.

I.e. Echo works just fine, but Blink and Pizza do not.

gigafied, 2017-01-27:
Also, oddly from Android seems to work so this seems to only affect `iOS/OSX (Central) > Linux (Peripheral)`

PalmerAL, 2017-01-29:
The code above seems to work fine with the descriptor removed. Thank you for the suggestion!

Also, having multiple characteristics works fine for me, as long as none of them use a descriptor.

gigafied, 2017-01-29:
@PalmerAL Try adjusting max mtu to 23.

I submitted a PR to make this easy: https://github.com/sandeepmistry/bleno/pull/266

savashito, 2017-02-08:
I have a similar problem, using the code snippet from @gigafied . When I run bleno on my OSX, and then connect my android device, I get the primary service uuid (fffffffffffffffffffffffffffffff0) and everything works. But when I run the same code from my raspberry pi, and scan with my phone, I don't get the primary uuid and instead get the two following uuids:

00001800-0000-1000-8000-00805f9b34fb
00001801-0000-1000-8000-00805f9b34fb

Any ideas?

sandeepmistry, 2017-02-13:
@savashito  please see: https://github.com/sandeepmistry/bleno#linux

savashito, 2017-02-14:
Thank you @sandeepmistry for your response. What in specific are you talking about from the linux installation? 
My bleno installation in linux  appears to work fine when communication with IOS or another machine running noble. But as I mention in my previous post it fails to list the services in Android. If I run it in MacOS, android is available to list the services and connect.

davidsi, 2017-02-22:
I have a related issue.

I can write to characteristics but I can not read from them.  All of my characteristics are similar to:

var MemoryCharacteristic = function() {

    MemoryCharacteristic.super_.call( this, {
        uuid: BLE.UUIDS.MemoryCharacteristic,
        properties: ['read'],
        descriptors : [ new bleno.Descriptor({
            uuid: BLE.UUIDS.BlenoDescriptor,
            value: 'gives information on the memory on the device'
        })]        
    });

    this._value = new Buffer(0);
};

MemoryCharacteristic.prototype.onReadRequest = function( offset, callback ) {

    console.log( "MemoryCharacteristic onReadRequest" );

    if( !offset ) {
        this._value = new Buffer(JSON.stringify({ 'freeMemory' : os.freemem(), 'totalMemory' : os.totalmem() }));
    }

    console.log('... value = ' + this._value.slice( offset, offset + bleno.mtu).toString() );

    callback( this.RESULT_SUCCESS, this._value.slice( offset, this._value.length ));
};

util.inherits( MemoryCharacteristic, BlenoCharacteristic );
module.exports = MemoryCharacteristic;

sandeepmistry, 2017-04-24:
Closing, as the original topic is most likely related to `bluetoothd` still running.

@davidsi please open a new issue, as your problem seems unrelated to the original topic.

closed 2017-04-24 by sandeepmistry