noble/bleno #267: Change a characteristic and notify (question)

cryptax, 2017-01-31:
Hi,

I am trying to modify the value of characteristic and **notify of its new value**. I fail to get it right. Can somebody help? _This is not a bug of bleno_, but a lack of understanding how to set up notify using bleno.

I have a piece of code that reads input from console (not sure I'm doing it the best way, but it works) and then calls `motorStateWrite` to change the value of a characteristic.
In `motorStateWrite`, I modify the value of my characteristic. And then, I'd like to notify there's a new value.
To my understanding, I should call `updateValueCallback(data)`. But it does not work: `TypeError: this.updateValueCallback is not a function` because it does not know that callback... and **I don't know how to create/supply it.**

More details if needed. First, I modify the data, call onWriteRequest for my characteristic:
```
myService.characteristics[2].onWriteRequest(new Buffer('01','hex'));
```

- I do not specify an offset nor callback - mainly because I don't know what I should supply there... ;)
- `characteristics[2]` corresponds to my motor state characteritic. It would be nicer to get the characteristic by uuid, but my JavaScripts skills are more than rusty :(

```
  myService.super_.call(this, {
	uuid: 'xxx',
	characteristics: [ new bleno.Characteristic({ // 0
	                                 uuid: 'xxx',
	                                 properties: ['read'],
	                                 value : new Buffer('02', 'hex')  
	}),
			   new bleno.Characteristic({ // 1
	                       uuid: 'xxxxxxxxxxxxxx'
	                       properties: ['read'],
	                       value : new Buffer('06', 'hex')  
			   }),
			    new bleno.Characteristic({ // 2
	                        uuid: 'yyyyyy',
	                        properties: ['read', 'notify'], 
	                        onReadRequest : motorStateRead,
				onWriteRequest : motorStateWrite,
				onSubscribe: motorStateSubscribe
			    }),
```

Then, my onWriteRequest actually goes to `motorStateWrite` and that's where I get the error:

```
function motorStateWrite(data, offset, withoutResponse, callback) {
    if (data.length !== 1) {
	console.log('[-] Failed to write motor state - bad data length: data=',data);
	callback(this.RESULT_INVALID_ATTRIBUTE_LENGTH);
    }
    else {
	motorState = data.readUInt8(0); // data = <Buffer 01>, motorState = 1
	this.updateValueCallback(data);
	callback(this.RESULT_SUCCESS);
    }
}
```

sandeepmistry, 2017-02-06:
It looks like you are missing the `write` property:

```
properties: ['read', 'notify'], 
```

needs to be

```
properties: ['read', 'write',  'notify'], 
```

closed 2017-02-06 by sandeepmistry

cryptax, 2017-02-07:
Hi @sandeepmistry ,
This does not really solve the issue. Indeed adding 'write' declares `motorStateWrite` correctly, but still it does not work, I still get `TypeError: this.updateValueCallback is not a function`:

```
readline.js:991
            throw err;
            ^

TypeError: this.updateValueCallback is not a function
    at Characteristic.motorStateWrite [as onWriteRequest] (/data/myservice.js:184:7)
    at /data/main.js:55:40
    at Interface._onLine (readline.js:245:5)
    at Interface._line (readline.js:591:8)
    at Interface._ttyWrite (readline.js:870:14)
    at ReadStream.onkeypress (readline.js:126:10)
    at emitTwo (events.js:106:13)
    at ReadStream.emit (events.js:191:7)
    at emitKeys (internal/readline.js:403:14)
    at emitKeys.next (<anonymous>)
```

How am I meant to declare `updateValueCallback`?

Code excerpt:
```
new bleno.Characteristic({ // motor state
	                       uuid: 'yyyyyy',
	                       properties: ['read', 'write', 'notify'], 
	                       onReadRequest : motorStateRead,
			       onWriteRequest : motorStateWrite,
			       onSubscribe: motorStateSubscribe
			    }),

...
function motorStateWrite(data, offset, withoutResponse, callback) {
    if (data.length !== 1) {
	console.log('[-] Failed to write motor state - bad data length: data=',data, ' offset=',offset);
	callback(this.RESULT_INVALID_ATTRIBUTE_LENGTH);
    }
    else {
	motorState = data.readUInt8(0); // data = <Buffer 01>, motorState = 1
	if (motorState == 0)
	    console.log("[+] Turning off");
	else
	    console.log("[+] Turning on");
	this.updateValueCallback(data);
	callback(this.RESULT_SUCCESS);
    }
}

function motorStateSubscribe(maxValueSize, updateValueCallback) {
    console.log("motorStateSubscribe()");
}
```
Kind regards.

sandeepmistry, 2017-02-13:
> How am I meant to declare updateValueCallback?

Please see the echo example: https://github.com/sandeepmistry/bleno/blob/master/examples/echo/characteristic.js#L40-L50

cryptax, 2017-02-14:
> master/examples/echo/characteristic.js#L40-L50

Ok, thanks.
That's the way to do it indeed.
Actually, my own code is not working currently, but it's another error,
I've got to investigate...