noble/bleno #275: Events on Characteristics

joegoldbeck, 2017-02-26:
I notice that Characteristics extend EventEmitter and emit events which correspond to the methods: `subscribe`, `readRequest` etc... I would like to use this as a feature, but see that it is not a documented part of the API. In particular, I would like to do `characteristic.once('subscribe', setUpListenerToProvideUpdates)`, so that I can set up this listener only once, and wait until there is a subscriber who wants the information it provides before setting it up. 

Do you expect this usage would be stable?

Thanks @sandeepmistry for the great module!

sandeepmistry, 2017-03-13:
@joegoldbeck good question.

> Do you expect this usage would be stable?

No guarantees on this, but it's unlikely to change in the near future. You could always use the callback and have an internal emitter.

closed 2017-03-13 by sandeepmistry

joegoldbeck, 2017-03-13:
Thanks for the response @sandeepmistry! Much appreciated