noble/bleno #280: Sending and Receiving between Raspberry pi 3 and Android

newbielinuxuser, 2017-03-15:
I've install Bleno in my raspberry pi 3, currently im able to send some plain text from android to raspberry pi 3, so how do i send some string back to Android ? Because I've tested with the callback function of onWriteRequest under new characteristic doesn't work, or are there any ways to trigger the writing part?

echo.js
`onWriteRequest : function(data , offset, withoutResponse, callback) {
this.value = data;
...
if(this.value.toString == 'received_text_from_android') {
      // Received text, send back received status to android
        }
}`

storm1er, 2017-03-15:
same problem, if someone can give us a real ping pong example, that could be awesome :)

newbielinuxuser, 2017-03-16:
Agree with the ping pong example, would be glad if there's sample about it =D

newbielinuxuser, 2017-03-21:
@storm1er you found the solution ?

storm1er, 2017-03-21:
Nope, I stop working on it for a time.
We just 'restart' the project with a friend, still no solution :( RPi seems ok but app still crashing on writeRequest

WayneKeenan, 2017-03-21:
Perhaps taking a look at the Echo example will help you?

https://github.com/sandeepmistry/bleno/blob/master/examples/echo/characteristic.js

newbielinuxuser, 2017-03-21:
@storm1er so your RPi(bleno) can do write request??

newbielinuxuser, 2017-03-21:
@WayneKeenan thanks, will try to explore into that

newbielinuxuser, 2017-03-29:
@storm1er noted, couldn't get it work yet, temporary KIV for this. I take it as April 5 ? cuz March 5 is like ... 3 weeks ago lol

storm1er, 2017-03-30:
@newbielinuxuser yup, April obviously, sry ...

storm1er, 2017-04-08:
Hey guys, great news, i'm close to the end \o/
I'm able to search, connect & read =)
I just did soe test on write, no crash as i had before. Have to do some cleaning & comment and i'll send a PR there.

ETA : 1 day

storm1er, 2017-04-09:
Hey !!!

using https://github.com/sandeepmistry/bleno on RPi : 
setService : https://github.com/storm1er/323-Rpi/blob/master/rpi/index.js#L23-L30
onReadRequest / on WriteRequest : https://github.com/storm1er/323-Rpi/blob/master/rpi/characteristics/relais_1.js#L26-L41

using https://github.com/evothings/cordova-ble on Android :
ReadCharacteristics : https://github.com/storm1er/323-Rpi/blob/master/android/libs/app.js#L462-L499
WriteCharacteristics : https://github.com/storm1er/323-Rpi/blob/master/android/libs/app.js#L176-L183

I have to comments more as ui text is in French.
If you want to ask somethings, i'm here =)

newbielinuxuser, 2017-04-09:
holy!! dam, you did it! trying these out =D 
will update you a.s.a.p

sandeepmistry, 2017-04-10:
Closing this as resolved. It would be great if someone could open a pull request for a Read + Write example.

closed 2017-04-10 by sandeepmistry

storm1er, 2017-04-10:
@sandeepmistry I would like to do it, but as it depends on two repo, i don't really know how to do it "the clean way"

I thought about put an new folder in example folder, then put a link to the cordova-ble/example folder, is that good ?

sandeepmistry, 2017-04-16:
@storm1er just a simple read write example in a new folder from "scratch" would be great.

storm1er, 2017-05-01:
@sandeepmistry I'll do it ;)
ETA : 1 week (05.08.2017)

houcem22, 2018-06-02:
hello everyone  :). @newbielinuxuser :  Can you please give me a link for a how-to send a string from android to raspberry pi 3. Thank you.