noble/bleno #281: stopAdvertising callback is never called

manthis, 2017-03-17:
I'm trying to get a confirmation that stopAdvertising works as expected. However, the callback is never called in my code:

> bleno.stopAdvertising(function() {
    console.log('Advertising stopped!');
});

I have also tried using the "on" event:

> bleno.on('advertisingStop', function() {
   console.log('Stopping advertising...');
});

But it also doesn't work. I'm on a Raspberry Pi 3 running last version of node.

leviathan88, 2017-03-28:
if and when you want to stop advertising you simply call
`bleno.stopAdvertising()`
you can also have a listener
```
bleno.on('advertisingStop', ()=> {
  console.log('The bleno.stopAdvertising() has just been called")
})
```

manthis, 2017-04-05:
Got you. Thank you very much.

closed 2017-04-05 by manthis

sandeepmistry, 2017-04-17:
@manthis are you still seeing this?

I just tried adding to the `test.js` in this repo and the callback was called. Tested on macOS and a Raspberry Pi 3.

```
setTimeout(function() {
  bleno.stopAdvertising(function() {
    console.log('Advertising stopped!');
  });

}, 5000);
```