noble/bleno #282: Setcap not working with Electron - need to use sudo

shekit, 2017-03-18:
Hi, I'm using bleno in an electron app and have followed all the steps for installation.

I launch my application using `electron main.js` 

As such I have run this step:
```sh
sudo setcap cap_net_raw+eip $(eval readlink -f `which electron`)
```

However I still get the error suggesting I run it with sudo. It works fine when I run it with sudo but it messes with other things in my application. I've applied the above step to both electron and node. Any ideas?

sandeepmistry, 2017-03-27:
@shekit can you run `getcap` on the `electron` binary?

It should print something like this:

```
pi@raspberrypi:~ $ getcap /usr/bin/nodejs
/usr/bin/nodejs = cap_net_raw+eip
```

shekit, 2017-04-03:
`$(eval readlink -f `which electron`)` returns `/usr/lib/node_modules/electron/cli.js`

If I run getcap on `$(eval readlink -f which electron)` I get this as the output:

`/usr/lib/node_modules/electron/cli.js = cap_net_raw+eip`

sandeepmistry, 2017-04-10:
You'll need to find the real binary electron runs, not just a .js file, then run setcap on the binary.

How about you run your Electron app, and then run `ps -eaf` to see what binaries are launched?

AnthonyMyatt, 2017-10-04:
Just in case anyone else comes to this issue looking for answers I believe it is the main Electron executable that needs the setcap command run for it, which would make the command as follows.
`sudo setcap cap_net_raw+eip /usr/lib/node_modules/electron/bin/electron`

elovin, 2017-11-02:
It's correct that you need to set the permissions on the electron executable, but this is probably the local electron executable since this is the one you will package and ship.
Also you need to add the packaged libnode.so to the dynamic linker in linux because the node version might differ from system to system or might not be installed at all.

This is what I did: sudo setcap cap_net_raw+eip ./node_modules/electron/dist/electron and than adding the absolute path to the dist directory to a new config file "electron-node.conf" in /etc/ld.so.conf.d/ after that I reloaded the linker with sudo ldconfig.

sandeepmistry, 2018-01-29:
@AnthonyMyatt thanks for the answer in https://github.com/sandeepmistry/bleno/issues/282#issuecomment-334100845.

closed 2018-01-29 by sandeepmistry